raspberrypi/pico-examples #2: Configuration error when follow the guide of MS Visual Studio Code setup.

xiongyu0523, 2021-01-23:
When I open the pico-example folder, the configuration process generate following error. How can I get sdk automatically download there?. It seems same for me when using command line instead of vsc, after type nmake

```
[main] Configuring folder: pico-examples 
[driver] Removing c:/Users/yuxiong/Documents/GitHub/pico-examples/build/CMakeCache.txt
[driver] Removing c:\Users\yuxiong\Documents\GitHub\pico-examples\build\CMakeFiles
[proc] Executing command: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/Users/yuxiong/Documents/GitHub/pico-examples -Bc:/Users/yuxiong/Documents/GitHub/pico-examples/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] Using PICO_SDK_PATH from environment (' ..\..\pico-sdk')
[cmake] CMake Error at pico_sdk_import.cmake:54 (message):
[cmake]   Directory 'C:/Users/yuxiong/Documents/GitHub/pico-examples/build/pico-sdk'
[cmake]   not found
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:4 (include)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!

```

xiongyu0523, 2021-01-23:
1) oh, the cmake configuration generate pico_sdk instead of pico-sdk under build. is it a bug?
2) is it necessary to use command line to generate cmake caches first and then open the pico-example by vsc? or can vsc do all thing at once?
3) it looks like when I type 'nmake' the command line is using MSVC compiler instead of arm-none-eabi-., what's wrong with the build system?

mrfaptastic, 2021-01-23:
Try setting PICO_SDK_PATH  to a non-relative path. i.e. "C:\users\.....\pico-sdk" and see if that helps?

I can't build a single example on windows either.

lurch, 2021-01-24:
Despite the similar names, [Visual Studio](https://visualstudio.microsoft.com/) and [VSCode](https://code.visualstudio.com/) are quite different things :wink:  The instructions in the Pico documentation were written for VSCode.

xiongyu0523, 2021-01-24:
Oh, sure I know it's for VSCode.  I just reuse the cmake, arm gcc installed in vs2019 folder, and make sure PATH is correct

lurch, 2021-01-24:
I've just spotted the ` -G Ninja` at the end your command-line. Does switching to `nmake` instead help? https://github.com/raspberrypi/pico-sdk/issues/5

xiongyu0523, 2021-02-01:
It seems it is not the same issue. The problem for me is, under build, there is no pico_sdk folder

```
[variant] Loaded new set of variants
[kit] Successfully loaded 4 kits from C:\Users\yuxiong\AppData\Local\CMakeTools\cmake-tools-kits.json
[main] Configuring folder: pico-examples 
[proc] Executing command: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/Users/yuxiong/Documents/GitHub/pico-examples -Bc:/Users/yuxiong/Documents/GitHub/pico-examples/build -G "NMake Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] Using PICO_SDK_PATH from environment (' ..\..\pico_sdk')
[cmake] CMake Error at pico_sdk_import.cmake:54 (message):
[cmake]   Directory 'C:/Users/yuxiong/Documents/GitHub/pico-examples/build/pico_sdk'
[cmake]   not found
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:4 (include)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
```

lurch, 2021-02-01:
Looks like you've set your `PICO_SDK_PATH` to a path that doesn't exist?
```
[cmake] Using PICO_SDK_PATH from environment (' ..\..\pico_sdk')
[cmake] CMake Error at pico_sdk_import.cmake:54 (message):
[cmake]   Directory 'C:/Users/yuxiong/Documents/GitHub/pico-examples/build/pico_sdk'
[cmake]   not found
```
Might be easiest to just download/clone [pico-sdk](https://github.com/raspberrypi/pico-sdk) to e.g. `C:/Users/yuxiong/Documents/GitHub/pico-sdk` and then set your `PICO_SDK_PATH` to that?

If that still fails, start again from scratch and carefully follow the instructions in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf ?

xiongyu0523, 2021-02-03:
I have pico-sdk cloned to `C:/Users/yuxiong/Documents/GitHub/pico-sdk` and pico-examples cloned to `C:/Users/yuxiong/Documents/GitHub/pico-examples`. And PICO_SDK_PATH is set to `..\..\pico-sdk` in Visual Studio Code. 

From error message `Directory 'C:/Users/yuxiong/Documents/GitHub/pico-examples/build/pico-sdk'`, it seems the ${PICO_SDK_PATH} was expanded to a wrong place. When I use command line to build, this ${PICO_SDK_PATH} is pointed to `'C:/Users/yuxiong/Documents/GitHub/pico-sdk`, which is correct. 

BTW, i have set the right env. 
[image]

I try to go through the chapter 8.2 Building on MS Windows again but could not find proper information.

lurch, 2021-02-03:
Thanks for confirming that the command-line build is working :+1: 

Just to check: are you definitely opening Visual Studio Code from inside a Developer Command Prompt, rather than from the Desktop or Start Menu?

xiongyu0523, 2021-02-06:
1) Command line has some other problem which seems to use X86 compiler to build my project and cause failures after `nmake`. I confirmed I have arm gcc, cmake and visual studio build tools installed and in PATH.  Let me copy the full log to you. 

2) yes. I open vscode by type `code` in VS2019 developer command line

```
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.8.4
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************

C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise>cd C:\Users\yuxiong\Documents\GitHub\pico-examples

C:\Users\yuxiong\Documents\GitHub\pico-examples>mkdir build

C:\Users\yuxiong\Documents\GitHub\pico-examples>cd build

C:\Users\yuxiong\Documents\GitHub\pico-examples\build>cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('..\..\pico-sdk')
Directory 'C:/Users/yuxiong/Documents/GitHub/pico-sdk' is found
Pico SDK is located at C:/Users/yuxiong/Documents/GitHub/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 8.2.1
-- The CXX compiler identification is GNU 8.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/Linux/gcc_arm/bin/arm-none-eabi-gcc.exe
Defaulting PICO target board to pico since not specified.
Using board configuration from C:/Users/yuxiong/Documents/GitHub/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Program Files/WindowsApps/PythonSoftwareFoundation.Python.3.8_3.8.2032.0_x64__qbz5n2kfra8p0/python3.8.exe (found version "3.8.7") found components: Interpreter
TinyUSB available at C:/Users/yuxiong/Documents/GitHub/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Found Doxygen: C:/Program Files/doxygen/bin/doxygen.exe (found version "1.8.17 (b5fa3cd1c6e6240e20d3b80a70e3f04040b32021*)") found components: doxygen missing components: dot
ELF2UF2 will need to be built
PIOASM will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/yuxiong/Documents/GitHub/pico-examples/build

C:\Users\yuxiong\Documents\GitHub\pico-examples\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.28.29336.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Scanning dependencies of target bs2_default
[  0%] Building ASM object pico_sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/boot2_w25q080.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target bs2_default_bin
[  0%] Generating bs2_default.bin
[  0%] Built target bs2_default_bin
Scanning dependencies of target bs2_default_padded_checksummed_asm
[  0%] Generating bs2_default_padded_checksummed.S
[  0%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target ELF2UF2Build
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is MSVC 19.28.29336.0
-- The CXX compiler identification is MSVC 19.28.29336.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.28.29333/bin/Hostx86/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
[... 53 lines cut ...]
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_double/double_init_rom.c.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/yuxiong/Documents/GitHub/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
[... cut ...]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Linux\gcc_arm\bin\arm-none-eabi-objcopy.exe:blink.hex: bad value
NMAKE : fatal error U1077: 'echo' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.28.29333\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```

lurch, 2021-02-06:
> C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\Linux\gcc_arm\bin\arm-none-eabi-objcopy.exe: blink.hex 64-bit address 0x4b4fa310000000 out of range for Intel Hex file

I'm not very familiar with the internals of compiler toolchains, but reading through https://bugs.launchpad.net/gcc-arm-embedded/+bug/1810274 it sounds like some versions of GCC might have problems when the 32-bit toolchain is being run on a 64-bit OS?

Could you try using the ARM GCC compiler we recommend in section 8.2.1 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf and see if that works any better than the ARM GCC included as part of "Microsoft Visual Studio 2019" ? (if that fixes the problem, we'll obviously add a suitable warning to the documentation.)

lurch, 2021-02-06:
> `-- The C compiler identification is GNU 8.2.1`

> Could you try using the ARM GCC compiler we recommend in section 8.2.1 of ...

Haha, funny coincidence :laughing:

xiongyu0523, 2021-02-07:
the arm gcc included in visual studio 2019 is 8.2.1

```
C:\Users\yuxiong>arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/microsoft\ visual\ studio/2019/enterprise/linux/gcc_arm/bin/../lib/gcc/arm-none-eabi/8.2.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw --libexecdir=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/lib --infodir=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/install-mingw/arm-none-eabi --with-libiconv-prefix=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-gmp=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-mpfr=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-mpc=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-isl=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-libelf=/tmp/jenkins/jenkins-GCC-8-build_toolchain_docker-519_20181216_1544945247/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for Arm Embedded Processors 8-2018-q4-major' --with-multilib-list=rmprofile
Thread model: single
gcc version 8.2.1 20181213 (release) [gcc-8-branch revision 267074] (GNU Tools for Arm Embedded Processors 8-2018-q4-major)
```

lurch, 2021-02-07:
Perhaps you misunderstood my last message? Section 8.2.1 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf (on page 35) links to https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads which currently has "Version 10-2020-q4-major" (GCC 10.2) available for download. Much newer than the version from 2018 that Visual Studio 2019 includes (and which hopefully fixes the objcopy bug you've encountered? :crossed_fingers: ).

f00lishMan, 2021-02-14:
I recently got everything working on vscode with cmake, pico, etc.

What I did:

1. deleted everything in .../pico/Downloads/pico-examples/build, deleted any ../pico/Downloads/pico-examples/.vscode folders to try to start fresh
2. checked that cmake/nmake works on just the terminal (Developer Command Prompt for VS 2019). i.e `cmake -G "NMake Makefiles" ..`, then `nmake`
3. Opened vscode via the VS 2019 dev cmd (I found that if executing `code` didn't work, I think I had previously installed VS 2019 _first_ months ago, so I uninstalled that and build tools, just installed build tools, and that worked) (Also ensure CMake Tools from Microsoft is installed in vscode)
4. Reopened the .../pico/Downloads/pico-examples folder, as in got rid of that folder via File->Close Folder and go back and File->Open Folder
5. In my c_cpp_properties.json for the folder, I have 

```
{
    "configurations": [
        {
            "name": "Pico",
            "compileCommands": "${workspaceFolder}\\build\\compile_commands.json"
        }
    ],
    "version": 4
}
```

This is very nice in that you won't have intellisense stuck saying it can't find files and can't resolve references, at least for me.  And you don't have to manually specify what compiler you're using, what defines you're using, etc.  This is the only configuration in my folder.
6. In my settings.json for the folder, I have included the setting `"cmake.generator": "NMake Makefiles"`.  This is to prevent vscode from trying to use ninja.  I had this problem when I would try to clean and rebuild via vscode.  This setting is part of the CMake Tools extension.
7. Selected the "active kit" on the bottom left, set that to "No Active Kit".  This is because the cmake files already specified the sdk, the toolchain, etc, so there's no need for vscode to pass extra arguments to the cmake command saying what compiler they're using.
[image]
8. Opened command palette (ctrl+shift+p), did `CMake: Configure`.  This was my output:

```
[main] Configuring folder: pico-examples 
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Release -Hc:/pico/Downloads/pico-examples -Bc:/pico/Downloads/pico-examples/build -G "NMake Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] PICO_SDK_PATH is C:/pico/Downloads/pico-sdk
[cmake] PICO platform is rp2040.
[cmake] PICO compiler is 
[cmake] PICO_GCC_TRIPLE defaulted to arm-none-eabi
[cmake] PICO target board is pico.
[cmake] Using board configuration from C:/pico/Downloads/pico-sdk/src/boards/include/boards/pico.h
[cmake] TinyUSB available at C:/pico/Downloads/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
[cmake] ELF2UF2 will need to be built
[cmake] PIOASM will need to be built
[cmake] -- Configuring done
[cmake] -- Generating done
[cmake] -- Build files have been written to: C:/pico/Downloads/pico-examples/build
```

9. Made sure my build target said `[all]`, Opened command palette (ctrl+shift+p), did `CMake: Clean Rebuild`.  Build went w/o issue.

And I was able to transfer .uf2 files over to the pico, they worked.  Reference tracing works w/o issue.  Able to clean and rebuild individual targets.  If it complains it can't find that compile_commands.json, it's because it was deleted in the clean, just reconfigure.

aallan, 2021-02-14:
> In my `c_cpp_properties.json` for the folder, I have
> ```
>{
>     "configurations": [
>         {
>             "name": "Pico",
>             "compileCommands": "${workspaceFolder}\\build\\compile_commands.json"
>         }
>     ],
>     "version": 4
> }
>```

This seems like a good addition to the project generator @JamesH65 ?

lurch, 2021-02-18:
> This is to prevent vscode from trying to use ninja.

There was an error in one of the SDK's cmake files, which caused `ninja` builds to fail, but that has since been fixed :slightly_smiling_face: https://github.com/raspberrypi/pico-sdk/pull/49

xiongyu0523, 2021-03-01:
Thanks for all your kind help. Here is my update to this issue

1) After install ARM GCC 8.2.1, the command line build issue was gone. 
2) The VSCODE2019 issue with PATH was introduced by incorrectly leading space for PICO_SDK_PATH env.

[image]

closed 2021-03-01 by xiongyu0523