raspberrypi/pico-examples #3: [build] ninja: error: 'pioasm/pioasm.exe', needed by 'dma/channel_irq

mrfaptastic, 2021-01-23:
Hi,

I've followed the "Getting started with Raspberry Pi Pico" pdf to the step, however attempting to compile the examples within Microsoft Visual Studio Code produces this error:

```
[main] Building folder: pico-examples 
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --build c:/Users/Anonymous/Documents/Hardware/PiPico/pico-examples/build --config Debug --target all -- -j 6
[build] ninja: error: 'pioasm/pioasm.exe', needed by 'dma/channel_irq/CMakeFiles/dma_channel_irq_pio_serialiser_pio_h', missing and no known rule to make it
[build] Build finished with exit code 1
```

mrfaptastic, 2021-01-23:
Also seems to have been raised here: https://www.raspberrypi.org/forums/viewtopic.php?t=299887&p=1802591

lurch, 2021-01-24:
Does this help? https://github.com/raspberrypi/pico-sdk/issues/5

Bazz-Man, 2021-01-24:
So I have VSC building the examples fine in Windows but only when I start VSC from a Developer Command Prompt and then type 'code', if I start VSC from its desktop Icon it doesnt build with the following error, even though the settings all look the same, what config am I missing to allow VSC to work from a normal startup from desktop icon

[main] Building folder: pico-examples 
[build] Starting build
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.EXE" --build c:/Users/barry/Downloads/pico-examples/build --config Debug --target blink --
[build] The system cannot find the file specified
[build] CMake Error: Generator: execution of make failed. Make command was: nmake /nologo blink && 
[build] Build finished with exit code 1

lurch, 2021-01-24:
Looks like maybe it can't find `nmake` in the path?

aallan, 2021-01-24:
Yup. It won't be able to find nmake if you start it from the desktop icon. I wrote the Windows instructions, and never found a way to get that working.

mrfaptastic, 2021-01-28:
Worked after trying again and ensuring I run vscode from the developer command prompt as @aallan advised.

closed 2021-01-28 by mrfaptastic