raspberrypi/pico-examples #4: hello_usb.c does not create USB-to-Serial port

xhawk18, 2021-01-23:
UF2 built from source code not works (not usb-to-serial port was found)

But the 'Hello World' UF2 file downloaded from works!
  https://www.raspberrypi.org/documentation/pico/getting-started/

xhawk18, 2021-01-23:
sorry, problem fixed!
we need download lib tinyusb by calling 
   git submodule update --init --recursive
in the pico-sdk.

closed 2021-01-23 by xhawk18

TheArchKnight, 2021-02-09:
I'm getting the same issue trying to use the mpu6050 i2c example, i have applied the command as well, and i have recompiled all of the examples. Any idea?

lurch, 2021-02-09:
@TheArchKnight I _think_ if you ran cmake before you'd cloned the tinyusb repo, that you need to delete your whole build directory and try again, so that the cmake build-caches get wiped too?

lurch, 2021-02-09:
Oh! And note also that by default the mpu6050 example isn't [configured](https://github.com/raspberrypi/pico-examples/blob/master/i2c/mpu6050_i2c/CMakeLists.txt) to use USB for stdio. Chapter 4 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf explains how to change that.

TheArchKnight, 2021-02-09:
@lurch it's working! it was a matter of enabling usb serial on the mpu6050 example, thanks for your help!