raspberrypi/pico-examples #6: Get an unique device ID

thibaultmeyer, 2021-01-23:
Hello,

is there a way to obtain a unique identifier for each device? Something like MAC address ?

Wren6991, 2021-01-23:
All RP2040s are identical, and there's no persistent state of any kind on-device, so there's no UID for RP2040s. Flash chips tend to have a unique identifier (usually the 4Bh command), so that would get you a per-board ID on Pico.

There's some careful footwork involved in sending serial commands to the flash chip, owing to the fact you are also executing code from it, so we can look at adding a general interface to the `hardware_flash` library for posting commands and getting a response back.

xhawk18, 2021-01-24:
unique id is the basic but necessary feature for all MCU ...

Wren6991, 2021-01-24:
> unique id is the basic but necessary feature for all MCU ...

Yes, it's also a standard feature on the flash chips we boot from, which have a 1:1 association with the MCU.

Small OTP stores take up a _lot_ of space on 40nm (it's the programming hardware, not the storage) so why make people pay for a duplicate feature on their silicon? We do need to add an example though, since as you say it needs to be something that's trivial to get at in user software. I'll get on that.

kripton, 2021-01-26:
Is the serial number the device presents to the USB host when running the bootloader also coming from the flash?
```
usb 1-4: Product: RP2 Boot
usb 1-4: Manufacturer: Raspberry Pi
usb 1-4: SerialNumber: E0C912D24340
```

XanaduNWH, 2021-01-27:
> Is the serial number the device presents to the USB host when running the bootloader also coming from the flash?
> 
> ```
> usb 1-4: Product: RP2 Boot
> usb 1-4: Manufacturer: Raspberry Pi
> usb 1-4: SerialNumber: E0C912D24340
> ```

I have a same SerialNumber in my dmesg :(

aallan, 2021-01-27:
> usb 1-4: SerialNumber: E0C912D24340

This is the firmware revision number. It will be the same on all chips.

kilograham, 2021-02-01:
merged into develop

closed 2021-02-01 by kilograham