raspberrypi/pico-examples #8: VS Code: Launching of gdb-multiarch not possible

schnappviech, 2021-01-23:
When setting a Raspberry Pi up as described in the Getting Started guide, two problems arise when launching the debugger:

- Setting "gdbpath" in launch.json is marked as not allowed
- When launching the setting is ignored, and arm-none-eabi-gdb is used, which is unavailable

It seems that setting "cortex-debug.gdbpath" can only be set in settings.json

edstevo, 2021-01-24:
Hi schnappviech. Please can you confirm the values for "cortex-debug.gdbpath" and or "cortex-debug.armToolchainPath"?

I am trying to launch the debugger, but am faced with the error:

"OpenOCD GDB executable "arm-none-eabi-gdb" was not found. Please configure "cortex-debug.armToolchainPath" correctly."

Thank you

schnappviech, 2021-01-24:
That is the same error I also got.

cortex-debug.armToolchainPath tells VS Code where to look for "arm-none-eabi-gdb", while cortex-debug.gdbPath specifies the name of the executable too.

Applying PR #9 did the trick for me. It requires gdb-multiarch to be in PATH though.

edstevo, 2021-01-24:
Oh man - I've been attacking this all day!!! Thank you so much. You're a _blinking_ legend!

lurch, 2021-01-26:
Fixed by #16

PR #16 (2021-01-26): Fix the case of the 'gdbPath' setting, and locate the SVD using $PICO_SDK_PATH
Fixes #8
closed 2021-01-27 by aallan