raspberrypi/pico-examples #10: Unknown CMake command "pico_generate_pio_header"

rwcooper33, 2021-01-24:
I get this error message trying to build the hello.pio example.

kilograham, 2021-01-24:
What steps did you use to try to build; what was the full output?

rwcooper33, 2021-01-24:
Here is what I did.

$ mkdir build

$ cd build

$ pwd
/home/rwcooper/pico/pico-examples/pio/hello_pio/build

$ cmake ..
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:3 (pico_generate_pio_header):
  Unknown CMake command "pico_generate_pio_header".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.13)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/rwcooper/pico/pico-examples/pio/hello_pio/build/CMakeFiles/CMakeOutput.log".

rwcooper33, 2021-01-24:
I realize what I did wrong. I was following the instructions on page 34 of the Pico C/C++ SDK document and was led astray. The CMakeFiles.txt at this level is not complete. In order to make it work the way i was doing it I would have to add additional instructions to the CMakeFiles.txt.

closed 2021-01-24 by rwcooper33

jackcampbell19, 2023-07-01:
For the next person who run into this issue here is an example of the other lines you need:
```
cmake_minimum_required(VERSION 3.12)
include(pico_sdk_import.cmake)

project(stepper_prog VERSION 1.0.0)

add_executable(stepper_prog)

target_sources(stepper_prog PRIVATE source.c)
target_link_libraries(stepper_prog PRIVATE pico_stdlib hardware_pio)

pico_sdk_init()

pico_generate_pio_header(stepper_prog ${CMAKE_CURRENT_LIST_DIR}/stepper.pio)

pico_add_extra_outputs(stepper_prog)
```

lurch, 2023-07-01:
> For the next person who run into this issue here is an example of the other lines you need:

This is all explained in Chapter 8 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?

jackcampbell19, 2023-07-01:
> > For the next person who run into this issue here is an example of the other lines you need:
> 
> This is all explained in Chapter 8 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf ?

@lurch Given the odd phrasing of your reply you are either:
1. Asking the question `Is this explained in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf?` to which the answer is: Yes! It is, specifically pages 30-31 :)
2. Or you are trying to be snarky. Clearly multiple people have been "led astray" (as the original poster puts it) by the [Pico C/C++ SDK Documentation](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) and I had personally been looking for an example to help me understand the full picture. Perhaps in the future you could phrase your replies in a more helpful way, such as `Full documentation for setting up a project can be found in Chapter 8 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf`.