raspberrypi/pico-examples #12: cmake does not respect PICO_SDK_PATH changing

cleverca22, 2021-01-24:
```
[nix-shell:~/apps/rpi/pico/pico-examples/build]$ cmake .. -DPICO_STDIO_USB=1
CMake Error at pico_sdk_import.cmake:54 (message):
  Directory '/nix/store/iakf0a4wi5al3a6gngf70arkmwcz7il1-source' not found
Call Stack (most recent call first):
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
See also "/home/clever/apps/rpi/pico/pico-examples/build/CMakeFiles/CMakeOutput.log".
See also "/home/clever/apps/rpi/pico/pico-examples/build/CMakeFiles/CMakeError.log".

[nix-shell:~/apps/rpi/pico/pico-examples/build]$ echo $PICO_SDK_PATH
/nix/store/60rlhylazjg06mpzn9qnpjwcs2n1ykh5-source

[nix-shell:~/apps/rpi/pico/pico-examples/build]$ rm CMakeCache.txt

[nix-shell:~/apps/rpi/pico/pico-examples/build]$ cmake .. -DPICO_STDIO_USB=1
Using PICO_SDK_PATH from environment ('/nix/store/60rlhylazjg06mpzn9qnpjwcs2n1ykh5-source')
Pico SDK is located at /nix/store/60rlhylazjg06mpzn9qnpjwcs2n1ykh5-source
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/1nkm42wc9rkpvly02q2vbd6gprjxsxjz-arm-none-eabi-stage-final-gcc-debug-wrapper-9.3.0/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /nix/store/60rlhylazjg06mpzn9qnpjwcs2n1ykh5-source/src/boards/include/boards/pico.h
-- Found Python3: /nix/store/bs03sg8b0gq2zr4v252hh9psp780qj5q-python3-3.8.5/bin/python3.8 (found version "3.8.5") found components: Interpreter
TinyUSB available at /nix/store/60rlhylazjg06mpzn9qnpjwcs2n1ykh5-source/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: /home/clever/apps/rpi/pico/pico-examples/build
```
after running a garbage collection (deleting the old `$PICO_SDK_PATH`) and updating `$PICO_SDK_PATH` to point to the new directory, cmake refused to respect that, and kept looking in the old path
deleting `CMakeCache.txt` made it forget and look in the new path, but it would be better if it detected the change automatically

lurch, 2021-01-24:
This seems like a bit of an edge-case - I'd imagine that for 99% of users the value of `$PICO_SDK_PATH` will never change? :shrug:

kilograham, 2021-01-24:
Yes, for now just delete your build directory and start again

kilograham, 2021-01-24:
Oh, i saw you fixed the problem, but were just pointing it out... cool yeah, I agree this should be fixed