raspberrypi/pico-examples #13: usb/device/dev_audio_headset not working as intended

techninja1008, 2021-01-24:
Any audio received from the device stutters and is distorted. After adding

shreekumar3d, 2021-02-02:
Doesn't work for me either.  For testing, I am playing audio using RhythmBox & recording using Audacity.

In my case, the recorded microphone audio is basically all noise.  I also find that the LED blinks rapidly all the time, which doesn't seem right.  According to the code, the LED must blink rapidly only when audio is being streamed.

shreekumar3d, 2021-02-02:
I tried a different approach to visualize what could be going wrong. I modified the sample to fill the microphone buffer with an incrementing integer.  Things seem to be working fine for about 7 seconds.  The sawtooth running from -1 to 1 being representative of that. But around 7 second mark, something happens.  I don't know what. Posting the image for reference.  Hoping it will be helpful.

[image]

lurch, 2021-02-02:
I _think_ this might be one of the examples that @liamfraser worked on? Perhaps he can give some clues.

techninja1008, 2021-03-03:
Is there anything I can do to help sort this out? USB isn't my area of expertise and I can't tell from either the example or the tinyusb code as to what is causing this.

liamfraser, 2021-03-04:
These examples are all copied from TinyUSB. Upstream TinyUSB has had some commits recently that should fix this. @ndinsmore might be able to confirm this. Longer term we will point at TinyUSB upstream instead of our own fork and work out a way of pointing at their examples from ours.