raspberrypi/pico-examples #19: USB stdio Read from host

Muny, 2021-01-27:
Hello,

There is an stdio example which shows how to print text to a host using printf; however I can't seem to figure out how to read data is that is being sent by the host to the RP2040? 

Perhaps this is unimplemented so far. If it isn't, I think there should be an example showing how to do it.

Muny, 2021-01-28:
I ended up yoinking some code from the rp2 port of micropython:

```
#include "tusb.h"

// Receive single character
int stdin_rx_chr(void) {
    for (;;) {
        if (tud_cdc_connected() && tud_cdc_available()) {
            uint8_t buf[1];
            uint32_t count = tud_cdc_read(buf, sizeof(buf));
            if (count) {
                return buf[0];
            }
        }
        tud_task();
    }
}
```

kilograham, 2021-01-31:
getchar() works!

closed 2021-02-01 by kilograham