raspberrypi/pico-examples #23: Raspberry Pi install  - error: invalid use of non-static data member ‘address_range::type

jomoengineer, 2021-01-29:
Following the install process for the Pico on a Raspberry Pi 4 using the pico_setup.sh script, when it got to the building of the pico-examples, a couple of failures were seen:

**OS Version: PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"**

##Build errors**
```
make[5]: *** [CMakeFiles/elf2uf2.dir/build.make:63: CMakeFiles/elf2uf2.dir/main.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:73: CMakeFiles/elf2uf2.dir/all] Error 2
make[3]: *** [Makefile:84: all] Error 2
make[2]: *** [blink/CMakeFiles/ELF2UF2Build.dir/build.make:111: blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1578: blink/CMakeFiles/ELF2UF2Build.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

## Build Output
```
Adding PICO_EXAMPLES_PATH to ~/.bashrc
Cloning https://github.com/raspberrypi/pico-extras.git
Cloning into 'pico-extras'...
remote: Enumerating objects: 151, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 151 (delta 12), reused 151 (delta 12), pack-reused 0
Receiving objects: 100% (151/151), 124.12 KiB | 1.88 MiB/s, done.
Resolving deltas: 100% (12/12), done.
Submodule 'lwip' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'lib/lwip'
Cloning into '/home/pi/development/pico/pico/pico-extras/lib/lwip'...
Submodule path 'lib/lwip': checked out 'c385f31076b27efb8ee37f00cb5568783a58f299'
Adding PICO_EXTRAS_PATH to ~/.bashrc
Cloning https://github.com/raspberrypi/pico-playground.git
Cloning into 'pico-playground'...
remote: Enumerating objects: 140, done.
remote: Counting objects: 100% (140/140), done.
remote: Compressing objects: 100% (117/117), done.
remote: Total 140 (delta 24), reused 128 (delta 17), pack-reused 0
Receiving objects: 100% (140/140), 1.86 MiB | 2.43 MiB/s, done.
Resolving deltas: 100% (24/24), done.
Adding PICO_PLAYGROUND_PATH to ~/.bashrc
Using PICO_SDK_PATH from environment ('/home/pi/development/pico/pico/pico-sdk')
Pico SDK is located at /home/pi/development/pico/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
Defaulting PICO target board to pico since not specified.
Using board configuration from /home/pi/development/pico/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.7 (found version "3.7.3") found components:  Interpreter
TinyUSB available at /home/pi/development/pico/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
ELF2UF2 will need to be built
PIOASM will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/development/pico/pico/pico-examples/build
Building blink
Scanning dependencies of target ELF2UF2Build
Scanning dependencies of target bs2_default
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] Building ASM object pico_sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/boot2_w25q080.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
Scanning dependencies of target bs2_default_bin
[  0%] No download step for 'ELF2UF2Build'
[  0%] Generating bs2_default.bin
[... 131 lines cut ...]
       ^
/usr/include/c++/5/bits/stl_vector.h:277:7: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/5/bits/stl_vector.h:264:7: note: candidate: std::vector<_Tp, _Alloc>::vector(const allocator_type&) [with _Tp = address_range; _Alloc = std::allocator<address_range>; std::vector<_Tp, _Alloc>::allocator_type = std::allocator<address_range>]
       vector(const allocator_type& __a) _GLIBCXX_NOEXCEPT
       ^
/usr/include/c++/5/bits/stl_vector.h:264:7: note:   candidate expects 1 argument, 2 provided
/usr/include/c++/5/bits/stl_vector.h:253:7: note: candidate: std::vector<_Tp, _Alloc>::vector() [with _Tp = address_range; _Alloc = std::allocator<address_range>]
       vector()
       ^
/usr/include/c++/5/bits/stl_vector.h:253:7: note:   candidate expects 0 arguments, 2 provided
/home/pi/development/pico/pico/pico-sdk/tools/elf2uf2/main.cpp: In function ‘int check_address_range(const address_ranges&, uint32_t, uint32_t, uint32_t, bool, address_range&)’:
/home/pi/development/pico/pico/pico-sdk/tools/elf2uf2/main.cpp:120:46: error: ‘address_range::type’ is not a class, namespace, or enumeration
             if (range.type == address_range::type::NO_CONTENTS && !uninitialize
                                              ^
/home/pi/development/pico/pico/pico-sdk/tools/elf2uf2/main.cpp: In function ‘int read_and_check_elf32_ph_entries(FILE*, const elf32_header&, const address_ranges&, std::map<unsigned int, std::vector<page_fragment> >&)’:
/home/pi/development/pico/pico/pico-sdk/tools/elf2uf2/main.cpp:153:51: error: ‘address_range::type’ is not a class, namespace, or enumeration
                     if (ar.type != address_range::type::CONTENTS) {
                                                   ^
make[5]: *** [CMakeFiles/elf2uf2.dir/build.make:63: CMakeFiles/elf2uf2.dir/main.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:73: CMakeFiles/elf2uf2.dir/all] Error 2
make[3]: *** [Makefile:84: all] Error 2
make[2]: *** [blink/CMakeFiles/ELF2UF2Build.dir/build.make:111: blink/elf2uf2/src/ELF2UF2Build-stamp/ELF2UF2Build-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1578: blink/CMakeFiles/ELF2UF2Build.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

aallan, 2021-01-29:
> -- The C compiler identification is GNU 5.5.0
> -- The CXX compiler identification is GNU 5.5.0

Your C compiler is very old (ancient!) and probably isn't capable of compiling the local tools, pioasm and elf2uf2, that are needed by the build system. Checking on the Raspberry Pi on my desk, which has a fresh OS install, the gcc version we're expecting to use is 8.3.0.

Please run,

```
$ sudo apt update
$ sudo apt full-upgrade
```

and try again. Alternatively if your C compiler is this out of date it might be easier to download a fresh copy of the OS onto a new SD Card and start again.

closed 2021-01-29 by aallan

jomoengineer, 2021-01-29:
The image is not that old considering it is a Pi 4 and I had recently performed a full upgrade so I am not sure where the gcc 5.x came from.
However, during the running of the pico-setup.sh script, I do believe it did download the "arm-linux-gnueabihf-gcc" tools.
'''
pi@raspberrypi:~ $ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
'''

Also, there was a gcc-8 version installed but it was not listed in the alternatives list.
```
pi@raspberrypi:~ $ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
```

After performing the suggestion from the following Ubuntu StackExchange post, I was able to get the gcc compiler to show version 8.3.0.
[https://askubuntu.com/questions/26498/how-to-choose-the-default-gcc-and-g-version](url)
```
pi@raspberrypi:~ $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc
```

The script is picking up the proper gcc now:
```
[  0%] Built target bs2_default_padded_checksummed_asm
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
```

lurch, 2021-01-30:
Yeah, when building something for Raspberry Pi Pico there's actually two separate copies of GCC needed - your native `gcc` which is used to compile `elf2uf2` and `pioasm` (which then later run on your Pi), and `arm-none-eabi-gcc` which compiles the binaries which run on the RP2040 (I believe that the 'none' indicates that there's no underlying OS?).
I _think_ that when you're running on a Pi that `arm-linux-gnueabihf-gcc` is the same thing as your native `gcc` ? (which is probably why your `arm-linux-gnueabihf-gcc` and `gcc-8` are reporting the identical version `Raspbian 8.3.0-6+rpi1` ? Whereas your `arm-none-eabi-gcc` is version 7.3.1)

In any case, I'm glad to see you managed to solve your problem by pointing your native `gcc` at a newer version :slightly_smiling_face: