raspberrypi/pico-examples #26: spi slave examples

Pillar1989, 2021-02-01:
Does it support spi slave ?  How is the speed? Do you have an example?

lurch, 2021-02-01:
Yes, it supports SPI slave mode (and also I2C slave mode). No, we don't have any examples yet.

Pillar1989, 2021-02-02:
spi slave is too important, I hope you can provide examples soon.

andylomas, 2021-03-12:
I'd like to support this request too!

I'm looking at using a Pico to collect input from a number of controls and pass that data to another microcontroller or Raspberry Pi. Having the Pico run SPI in slave mode seems like a natural way to quickly comminicate the data.

arhuaco, 2021-04-04:
Hello there. I have 2 picos around and I really need them to implement some sort of DMA that uses few pins of a Raspberry Pi. For this I need SPI slave, ideally. Since the RPI doesn't provide slave mode and it was removed in the 4th version, this example is really needed.

michaelstoops, 2021-04-05:
I am in the research phase for a project of my own, in which a Pico acts as a SPI slave. As I learn how to do it for myself, I may as well write an example for others to follow. If I write the example and send the pull request, would you accept it?

I'm thinking there would probably be a pair of programs: one for a SPI master which takes input from stdio and writes it to SPI bus as master. The other is for SPI slave, which reads bytes off the SPI bus and writes them to its own stdio in some form. Perhaps the master reads in bytes, echoes the byte, and the slave outputs the bits to stdio as ASCII and newline. So typing `pi` into the master causes the slave to output:
```
01110000
01101001
```
That sounds simple, good enough to make the point, and might actually be useful for some basic diagnostics.

Would such an example be a valuable code contribution?

michaelstoops, 2021-04-05:
Come to think of it, the example could probably run on one Pico , with SPI master and slave running on separate cores.

lurch, 2021-04-07:
> The other good news is that I think I will be working on emulating an SD card through PIO though, and I hear some people are interested in that.

See also https://github.com/raspberrypi/pico-extras (in case you've not already seen it).

michaelstoops, 2021-04-07:
> See also https://github.com/raspberrypi/pico-extras (in case you've not already seen it).

I have, thank you!

michaelstoops, 2021-04-12:
It turned out that I needed to write the SPI slave code after all, so here you go: https://github.com/raspberrypi/pico-examples/pull/101.

It turned out to be significantly more difficult than I expected. I didn't have enough experience with PIO to really know what to write or how to design it, and the existing SD master example was more complex than I wanted to start with.

So I took an incremental approach and attempted a basic SPI slave mode using the hardware SPI. I thought it would be easy but it was not. I derived my code from spi_flash, which bit-bangs the CS line and doesn't say why other than some `FIXME` tags in the code without further explanation.

It appear that the PrimeCell SSP in Motorola SPI mode appears to use the rising edge of the CS pin as a signal to latch the ISR into the receive FIFO, or something like that. A master can write all day long and never notice, but a slave that responds to commands won't read anything without that pulse.

Took a while to get it sorted, but the example works well. Hopefully others will have it easier because of my labors.

closed 2021-04-20 by Pillar1989