raspberrypi/pico-examples #29: pico-examples/pio/logic_analyser not working for some values of CAPTURE_PIN_COUNT

cmoulang, 2021-02-02:
The logic analyser example does not work where CAPTURE_PIN_COUNT does not divide exactly into 32.

For example with CAPTURE_PIN_COUNT=4 I get the following *correct* output:

`16: ----____________----____________----____________----____________----____________----____________`
`17: ------------____------------____------------____------------____------------____------------____`
`18: ________________________________________________________________________________________________`
`19: ________________________________________________________________________________________________`

However, with CAPTURE_PIN_COUNT=3 I get the following *incorrect* output:

`16: -----------____________________------------___________________------------___________________---`
`17: _______________----__------____-______________----___-----____--_____________----____----____---`
`18: ---_______-____------___________--____________-------___________-____________--------___________`

The attached file contains a fix.  
[logic_analyser.c.txt](https://github.com/raspberrypi/pico-examples/files/5910657/logic_analyser.c.txt)

In summary: 
- PIO discards bits when a sample overruns the available space in the ISR. `push_threshold` is set to a multiple of `pin_count` to prevent this.
- The capture_buf size calculation has been updated.
- Calculation of the word and bit offsets in print_capture_buf() fixed.

Wren6991, 2021-02-11:
Fixed, thank you!

closed 2021-02-11 by Wren6991

cmoulang, 2021-03-06:
Great! Thanks!