raspberrypi/pico-examples #31: Any sample for SSD1306 based OLED display?

pankaj-nikam, 2021-02-02:
Hi team, is there any sample for SSD1306 based OLED display? Can you please share? I am looking for 128x32 display.
Thanks.

lurch, 2021-02-02:
There's an SSD1306 MicroPython example [here](https://github.com/raspberrypi/pico-micropython-examples/tree/master/i2c/1306oled) but I dunno how easy/hard it would be to convert to C. There's some C I2C examples [here](https://github.com/raspberrypi/pico-examples/tree/master/i2c).

pankaj-nikam, 2021-02-03:
Yes, I found that sample and its working nice. I wanted C version because the Python currently does not support machine.RTC.

HeadBoffin, 2021-02-04:
The Python example will be driven by C under the hood so you could fish it out of the Micropython source code

mbober1, 2021-02-05:
I ported SSD1306 lib to RP2040. Feel free to use it.
https://github.com/mbober1/RPi-Pico-SSD1306-library

pankaj-nikam, 2021-02-06:
Thanks for the porting @mbober1 😃

lurch, 2021-02-09:
See also https://github.com/raspberrypi/pico-examples/pull/49 :slightly_smiling_face:

lurch, 2021-11-05:
I guess this was "fixed" by #130 ? That got merged into https://github.com/raspberrypi/pico-examples/releases/tag/sdk-1.3.0 so take a look at https://github.com/raspberrypi/pico-examples/tree/master/i2c/oled_i2c

mkerna, 2023-08-31:
There is an example in the repository at [.](https://github.com/raspberrypi/pico-examples/tree/master/i2c/ssd1306_i2c)  Please close this issue.