raspberrypi/pico-examples #33: PIO / WS2812 / parallel -- documentation?

henrygab, 2021-02-02:
First, this is a **_great_** example, showing how to implement scatter-gather list DMA functionality on this board, for a real-world scenario.

**Would pull requests for documentation be accepted, and if so, what is the preferred format?
(e.g., embedded in source, separate README.md, other?)**

For example, I think this directory should have at least a few sections documented:
1. WS2812B signaling (even if pointing to a section of the SDK)
2. What the parallel version does (up to 32(30) outputs simultaneously!)
3. How the data for parallel WS2812B sample is massaged into the bitplane format
4. How two DMA channels are used to implement scatter-gather support

I've given a strawman starting point for the last item below.

<details><summary>DMA Architecture Summary (click to expand)</summary><P/>

This sample implements scatter-gather list based transfers, using two chained DMA channels:

1.  The "main" DMA channel, which transfers a eight (8x) 32-bit values to the state machine.  Together, these values contain one byte of data for up to 32 output channels.

2. The "main" DMA channel is configured to only generate an interrupt when a trigger register is written with a nullptr/zero value.  Sweet!  Only one interrupt should ever generate, at the end of all transfers:
https://github.com/raspberrypi/pico-examples/blob/33854562cd08398c4b48bb1ed7fa022c2177076d/pio/ws2812/ws2812_parallel.c#L251

3. The "cb" (callback?) DMA channel writes a single 32-bit value to the main DMA channel's configuration register.  It uses the third alias, so that writing the updated read address also triggers the start of the main DMA channel.

4. The interrupt is enabled only for the main DMA channel, using `DMA_IRQ_0` 
https://github.com/raspberrypi/pico-examples/blob/33854562cd08398c4b48bb1ed7fa022c2177076d/pio/ws2812/ws2812_parallel.c#L269-L271

</details>

kilograham, 2021-02-02:
Absolutely documentation very much welcome... this was just thrown together by me for my own testing and figured it might help someone ;-0

certainly some in source comments are useful and/or tidying up of code. Also maybe a README.adoc (@aallan can make suggestions about what he prefers)

aallan, 2021-02-03:
Hey there! Yes, absolutely we'd love documentation PRs. If you take a look in the `pico-examples` repo you'll find that some of the directories have a `README.adoc`, 

```
$ find . -name "README.adoc" -print
./i2c/lcd_1602_i2c/README.adoc
./i2c/mpu6050_i2c/README.adoc
./spi/bme280_spi/README.adoc
./spi/mpu9250_spi/README.adoc
./gpio/dht_sensor/README.adoc
./gpio/hello_7segment/README.adoc
$
```

These `README.adoc` files are the preferred method for documenting example code. If we take a look at the `gpio/dht_sensor/` directory,

```
$ ls
total 1576
drwxr-xr-x  8 aa  staff   256B 20 Jan 23:41 ./
drwxr-xr-x  6 aa  staff   192B 20 Jan 23:41 ../
-rw-r--r--  1 aa  staff   175B 20 Jan 23:41 CMakeLists.txt
-rw-r--r--  1 aa  staff   2.1K 20 Jan 23:41 README.adoc
-rw-r--r--  1 aa  staff   2.1K 20 Jan 23:41 dht.c
-rw-r--r--  1 aa  staff    59K 20 Jan 23:41 pi-and-pico-uart-and-dht-sensor.fzz
-rw-r--r--  1 aa  staff   303K 20 Jan 23:41 pi-and-pico-uart-and-dht-sensor.png
-rw-r--r--  1 aa  staff   409K 20 Jan 23:41 serial-output.png
$
```

as well as the example code and CMake infrastructure you'll see a `README.adoc` file and accompany Fritzing and PNG files. 

These asciidoc files will render when looking at the code example [here in Github](https://github.com/raspberrypi/pico-examples/tree/master/gpio/dht_sensor) but will also get (automagically) imported into the PDF documentation, see "Appendix A: App Notes" in the [C/C++ SDK](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf) book, along with the example code.

The `[dht_sensor/README.adoc](https://github.com/raspberrypi/pico-examples/blob/master/gpio/dht_sensor/README.adoc)` is probably the best template example to copy. You can find the Fritzing part for Pico [here](https://datasheets.raspberrypi.org/pico/Pico-R3-Fritzing.fzpz) so that you can create a wiring diagram.

kilograham, 2021-02-03:
even without the fritzing, the documentation of how it works is invaluable!

aallan, 2021-02-03:
> even without the fritzing, the documentation of how it works is invaluable!

Oh, agreed!

henrygab, 2021-02-25:
I know that I won't be able to get to this.  I will close this now.

closed 2021-02-25 by henrygab