raspberrypi/pico-examples #34: flash_range_erase does not work

BlueCocoa, 2021-02-03:
Hi, I was trying to erase the internal flash following the example, but it seems that `flash_range_erase` does not work. The whole program just gets stuck after calling `flash_range_erase`. 

I want to erase out about 1MB area right after the first 1MB on the Pico Pi. Therefore the parameters I passed is `flash_range_erase(1048576, 1044480)`. And `1044480` is indeed a multiple of 4096 (i.e., the sector size), 1044480=255*4096.

My pico-sdk version is [2d5789eca89658a7f0a01e2d6010c0f254605d72](https://github.com/raspberrypi/pico-sdk/tree/2d5789eca89658a7f0a01e2d6010c0f254605d72).

My CMakeLists.txt is,
```
cmake_minimum_required(VERSION 3.12)

set(PICO_SDK_PATH "/home/ubuntu/src/pico-sdk")
# initialize the SDK based on PICO_SDK_PATH
# note: this must happen before project()
include(pico_sdk_import.cmake)

project(pico_fs C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# initialize the Pico SDK
pico_sdk_init()

include_directories(${CMAKE_CURRENT_BINARY_DIR})
include_directories("${PICO_SDK_PATH}")
add_executable(pico_test_flash main.c)

target_link_libraries(pico_test_flash
    pico_stdlib
    hardware_flash
)

# enable usb output, disable uart output
pico_enable_stdio_usb(pico_test_flash 1)
pico_enable_stdio_uart(pico_test_flash 0)

# create map/bin/hex file etc.
pico_add_extra_outputs(pico_test_flash)
```

The main.c file is
```
#include <stdio.h>
#include <stdlib.h>

#include "pico/stdlib.h"
#include "hardware/flash.h"

// We're going to erase and reprogram a region ~1MB from the start of flash.
// Once done, we can access this at XIP_BASE + 1MB.
#define FLASH_TARGET_OFFSET (1024 * 1024)

int main() {
    stdio_init_all();
    getchar();
    printf("Erasing target region...\n");
    flash_range_erase(FLASH_TARGET_OFFSET, 255 * FLASH_SECTOR_SIZE);
    printf("Done.\n");

    while (true) {
        int ret = getchar_timeout_us(1000 * 1000);
        if (ret != PICO_ERROR_TIMEOUT) {
            break;
        }
        printf("Press any key to exit\n");
    }
}
```

When I connected my Pico Pi using `screen` and pressed the enter key, the only output was `Erasing target region...`. I waited for about 15 minutes, there was still no more output. (Although occasionally the program works, most of the time it just hangs there)

Wren6991, 2021-02-03:
Thank you for providing the full modified CMakeLists.txt file -- my first guess is that this is caused by a USB IRQ for the `printf` firing whilst the flash programming is in progress, which then fetches invalid data from the XIP window.

Could you please confirm whether running `cmake -DPICO_COPY_TO_RAM=1 ..` in your build directory, and then rebuilding, fixes the symptom?

The correct fix is probably to either disable IRQs on either side of those function calls (which are documented in the header as being non-IRQ-safe, but clearly this is not helpful when the USB IRQ was installed in the background for you). Pulling the IRQ-related parts of the USB code into RAM is probably far too expensive.

BlueCocoa, 2021-02-03:
Hi @Wren6991 thank you for the reply. I guess it was the USB IRQ that caused the problem. 

Sorry that I didn't read the docs carefully enough and disabling USB IRQ before calling `flash_*` functions indeed solved my problem.

closed 2021-02-03 by BlueCocoa

lurch, 2021-02-03:
So I guess the workaround here is just using stdio over UART instead of stdio over USB? Or would that also run into similar IRQ problems?

kilograham, 2021-02-03:
the moral of the story is to be very careful using flash_range_erase etc. when running from flash.

nhhiep92, 2021-07-27:
Hello kilograham,
I am facing the issue with multi core when erasing flash. Do you have any suggestion for me?

ulri-me, 2023-01-02:
This issue/bug/behaviour took me many hours to debug as the issue appeared 19 of 20 times without any obvious reasons.

My solution was this blog post:
https://www.makermatrix.com/blog/read-and-write-data-with-the-pi-pico-onboard-flash/

It claims the usage of the code
uint32_t ints = save_and_disable_interrupts();
flash_range_erase(___, ___);
flash_range_program(___, ___, ___);
restore_interrupts (ints);

This works just fine.

And @kilograham:
In case you want to include this into a multicore environment then just scope the code skeleton above with a semaphore solution by sync library.