raspberrypi/pico-examples #35: Python SDK can't import array

PWok, 2021-02-03:
I updated my pico to use Micropython from rp2-pico-20210202-v1.14.uf2 (newest release as I'm writing) and after doing so some of my test programs stopped working. Im getting an ImportError: no module named 'array'. At first I thought that it was just removed, but now I also can't use rp2.asm_pio() getting similar error:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rp2.py", line 228, in asm_pio
  File "rp2.py", line 36, in __init__
ImportError: no module named 'array'

aallan, 2021-02-03:
Hey there! The RP2040 MicroPython port has now been merged upstream to the main MicroPython repo, see micropython/micropython#6791. Going forward you should report problems with the port  there, https://github.com/micropython/micropython/.

closed 2021-02-03 by aallan

lurch, 2021-02-03:
There's now an upstream bug for this: https://github.com/micropython/micropython/issues/6837

thijstriemstra, 2021-02-07:
> Hey there! The RP2040 MicroPython port has now been merged upstream to the main MicroPython repo, see micropython/micropython#6791. 

Wow. So all of the nice documentation for the pico is already outdated?

lurch, 2021-02-07:
> So all of the nice documentation for the pico is already outdated?

Huh? Can you point to a specific part of the documentation that is / will become outdated? Note that even though the repository for the source-code has changed from https://github.com/raspberrypi/micropython/ to https://github.com/micropython/micropython/ , this should be irrelevant to most users as we'll still be offering ready-compiled MicroPython UF2 files downloadable from https://www.raspberrypi.org/documentation/pico/getting-started/

(Also, we're actively updating the documentation available from https://www.raspberrypi.org/documentation/pico/getting-started/ - the PDFs are already up to V1.2 :slightly_smiling_face: )