raspberrypi/pico-examples #38: Referenced to alias 3 instead of alias 2.

AsafFisher, 2021-02-03:
https://github.com/raspberrypi/pico-examples/blob/33854562cd08398c4b48bb1ed7fa022c2177076d/dma/control_blocks/control_blocks.c#L27

henrygab, 2021-02-04:
Can you help me understand the question here?
The comment appears consistent with the data structure:

https://github.com/raspberrypi/pico-examples/blob/33854562cd08398c4b48bb1ed7fa022c2177076d/dma/control_blocks/control_blocks.c#L32-L37

The last of the four registers in each alias is special ... when written to (except when writing nullptr), it automatically starts the DMA.
Here, alias 3 is used, and the control channel writes the following two words at each transfer:

* alias3->TRANS_COUNT
* alias3->READ_ADDR

The write to alias3->READ_ADDR thus triggers the data channel to perform a DMA operation, using this new read address and byte count.

In essence, this little IoT device supports configurable [scatter/gather DMA](https://en.wikipedia.org/wiki/Vectored_I/O) ... fairly advanced capability for the price.

AsafFisher, 2021-02-04:
On alias 3 the length(trans_count) is not before the read address unlike what the comment says

Wren6991, 2021-02-04:
> On alias 3 the length(trans_count) is not before the read address unlike what the comment says

Yes it is. The comment in question:

```
// Alias 3:  CTRL        WRITE_ADDR    TRANS_COUNT   READ_ADDR 
```

The layout of the control block we write into it at an 8 byte offset (in the example code, which does work):

```
const struct {uint32_t len; const char *data;} control_blocks[] = {
```

The section of the struct header that defines this:

```
    io_rw_32 al3_ctrl;
    io_rw_32 al3_write_addr;
    io_rw_32 al3_transfer_count;
    io_rw_32 al3_read_addr_trig;
}  dma_channel_hw_t;
```

The table in the datasheet:

[image]

These all seem 100% consistent to me. If you find something disagrees with those 4 matching definitions, then please point me to it so I can fix it.

AsafFisher, 2021-02-04:
My mistake (:

closed 2021-02-04 by AsafFisher