raspberrypi/pico-examples #41: PIO / WS2812 ... Bug on preprocessor debug disable?

henrygab, 2021-02-05:
https://github.com/raspberrypi/pico-examples/blob/33854562cd08398c4b48bb1ed7fa022c2177076d/pio/ws2812/ws2812_parallel.c#L285-L291

First, I'm at a loss trying to understand why this is ever disabled, given that the PICO has dedicated SWD pins.  This also isn't checking against `PICO_DEFAULT_UART_TX_PIN`, `PICO_DEFAULT_UART_RX_PIN`, or `PICO_DEFAULT_LED_PIN` ... was it leftover from the FPGA days?

Even if it's OK for the non-parallel, however, it's not correct for this file, since it outputs to 2 consecutive GPIO.  `PIN_TX` is only the *first* GPIO that the state machines will push output through.

Thoughts appreciated?

lurch, 2021-02-05:
I suspect this _may_ be timing debug-code that was left in by accident, similar to https://github.com/raspberrypi/pico-sdk/issues/28 ?

henrygab, 2021-02-06:
Yep, that seems essentially the same.  :)  Please mark as duplicate or close, as per your preference!

lurch, 2021-02-06:
IMHO this _isn't_ a duplicate, because it's affecting a different source-file than was being discussed in https://github.com/raspberrypi/pico-sdk/issues/28

henrygab, 2021-02-08:
**You're right**.  The linked issue said it'd be going through all source, and [SDK commit #31](https://github.com/raspberrypi/pico-sdk/pull/31) closed #28.

My mistake was two-fold:
1. The linked issue said it would fix all the samples
2. I failed to realize that the linked issue was in a different depot

So, **_yes_**, this issue should remain alive in this depot, to either:
1. remove all instances of `CU_REGISTER_DEBUG_PINS()`, `CU_SELECT_DEBUG_PINS()`, `DEBUG_PINS_xxx`, etc.;  **_OR_**
2. comment out at least the `CU_REGISTER_DEBUG_PINS()`, `CU_SELECT_DEBUG_PINS()` lines

lurch, 2021-02-08:
@kilograham For your consideration...

kilograham, 2021-02-08:
just delete the debug pins stufff

lurch, 2021-02-08:
Fancy submitting a PR to the `develop` branch @henrygab ?

henrygab, 2021-02-09:
OK, see PR #47.

henrygab, 2021-02-12:
Manually closing because I failed to properly link the PR.  :)

PR #47 (2021-02-11): Fix #41 - Remove debug pin per @kilograham
Fix #41.
commit 8b016f0b (2021-02-11): Fix #41 - Remove debug pin per @kilograham (#47)
commit c63ef7cb (2021-03-05): Fix #41 - Remove debug pin per @kilograham (#47)
closed 2021-02-12 by henrygab