raspberrypi/pico-examples #50: WS2812 Diver code run through in raspberry-pi-pico-c-sdk doc.

Bleep42, 2021-02-10:
In raspberry-pi-pico-c-sdk doc, Section 3.2.2 Pg 37 on.
The Times specified in the code and on page 37, are T1=2, T2=5, T3=3 however the timing diagrams further down Fig 4 to 7 imply timings of T1=4 T2=3 T3=3 assuming I am reading them correctly. Also Fig 6 mentions T1 twice, I think the second mention should be T2.
Assuming I understand how the code works, I believe the actual timing values should be T1=2, T2=2, T3=3 which I believe should give 358ns for a '0' bit, 716ns for a '1' bit and a minimum inter bit spacing of 537ns. My understanding is that the '1' bit should be roughly twice the length of the '0' where as in the example it is over 3 times?  (2clk + 5clk) for a '1' or just 2clk for a'0'
Though I understand that the original timings will work fine, (they do) it doesn't help to make how the code works very clear, especially when looking at the timing specifications for the WS2812, I also understand that the specification timings  can vary quite a bit depending on where you look!

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for Raspberry Pi Pico Projects

psitech, 2021-02-23:
After scrutinizing the raspberry-pi-pico-c-sdk document and measuring with a logic analyzer, it boils down to the following:
- The statemachine [SM] is running at 8Mhz. Clock period is therefor 0.125 usec.
- The clock frequency for updating the WS2812 LEDs is 800kHz. Thus 1 bit time of the WS2812 LED is 1.25 usec.

So 1 bit time of the LED equals 10 clocks of the statemachine. These 10 SM clocks are used for generating either a low or high LED bit. An SM clock is indicated as a vertical dotted line in the image below.

[image]

As you might have guessed, (T1 + T2 + T3) must be 10.
From the WS2812B **[(link)](https://github.com/raspberrypi/pico-examples/files/6027755/WS2812B.pdf)** datasheet:
[image]
[image]

With the original T1=2, T2=5, T3=3: 
- a "0 code" is 2 SM clocks high and 8 (5+3) SM clocks low [0.250 usec high/1 usec low].
- a "1 code" is 7(2+5) SM clocks high and 3 SM clocks low [0.875 usec high/0.375 usec low].

Since the "0 code" timing is just on the edge of the specification, I changed T1 to 3 and T2 to 4. so now a "0 code" is 0.375 usec high/0.875 usec low. Now the timing is within spec.
I verified this to be correct using a logic analyzer running at a resolution of 2.5 nsec.

Regards,
Paul

Bleep42, 2021-02-24:
Hi Paul,
Firstly, my main comment is that the diagrams Fig 4 to 7, one of which you reproduce do not reflect the values used in the code, which to me is very confusing.  In the code T1=2, T2=5, T3=3 however the timing diagrams further down Fig 4 to 7 imply timings of T1=4 T2=3 T3=3, therefore the diagrams need to be corrected, to be the same as the code.

You imply that the total of T1+T2+T3 has to be 10 "As you might have guessed, (T1 + T2 + T3) must be 10." as far as I am aware this is not the case and I have used numerous values for T1,T2,T3 which do not add up to 10 and everything works perfectly; there is the possibility that the exact 800kHz won't be met, depending on what clocks/dividers are abe to be used, but it will usually be close enough for all practical purposes, for example I am using T1=2, T2=2, T3=3, which gives a 178nS split, giving 358ns for a '0' bit, 716ns for a '1' bit and a minimum inter bit spacing of 535ns. This works fine.
However the values used in the code of T1=2, T2=5, T3=3 are I suspect actually intended for the WS2812B, which has different timing requirements, so something needs correcting in the document and or the code.
Regards, Kevin.

psitech, 2021-02-24:
Hi Kevin,
Agree, the graphs got me confused too initially. And, as you suggest, a documentation update is indeed needed.

Interesting to hear from you that not-adding-up-to-10 for T1+T2+T3 is also working!
I got to the must-add-up-to-10 conclusion by looking at the Python code, which sets the SM clock to 8MHz:
_sm = rp2.StateMachine(0, ws2812, freq=8_000_000, sideset_base=Pin(PIN_NUM))_

Anyway, later today I will play with different Tx values [not necessarily adding up to 10] and see what the output shows on the logic analyzer. My guess is that these LEDs are somewhat forgiving with respect to timing.
Actually, the header **_Data transfer time( TH+TL=1.25µs±600ns)_** above the timing table is questionable. It sounds as if the bit time can be anywhere between 0.65 usec and 1.85 usec? This violates the subsequent values and tolerances...

Regards,
Paul

lurch, 2021-02-24:
> However the values used in the code of T1=2, T2=5, T3=3 are I suspect actually intended for the WS2812B, which has different timing requirements, so something needs correcting in the document and or the code.

Oh!! I can't speak for the author of the code you're talking about here, but personally I didn't realise that WS2812 and WS2812B were actually different, nor that they had different timing requirements. Confusingly, https://github.com/raspberrypi/pico-micropython-examples/tree/master/pio/neopixel_ring talks about both "NeoPixel" and "WS2812", whereas the example projects at the bottom of https://www.raspberrypi.org/documentation/pico/getting-started/ reference "NeoPixel" and "WS2812b" :confused:

psitech, 2021-02-24:
Hi Andrew, thanks for compiling the table. 
As you can see the tolerances are quite wide so one set of T1-T2-T3 should work on all of them. Even the older WS2811 should work with that setting.
More on the T1+T2+T3 != 10 later today.

Regards,
Paul

psitech, 2021-02-24:
Allright, have been playing with different combinations of T1-T2-T3.
For your information: I'm running this Python code: [NeoPixelRing_Rainbow.zip](https://github.com/raspberrypi/pico-examples/files/6037298/NeoPixelRing_Rainbow.zip) and using the Adafruit 24-LED neopixel ring [(link)](https://www.adafruit.com/product/1586).

**TL;DR** if you want 800kHz bit rate, T1+T2+T3 has to be 10 [with state machine clock of 8MHz].
**TL;DR** these LEDs are quite forgiving on the timing.

My conclusion is that 3-4-3 is the most safe T1-T2-T3 setting specification-wise for these Neopixels.

Regards,
Paul

Here are the logic analyzer outputs for different combinations of T1-T2-T3:

1-4-1 [ring did not display]
[image]

1-5-1 [ring did not display]
[image]

1-6-1 [ring displayed running rainbow]
[image]

1-8-1 [ring displayed running rainbow]
[image]

2-2-2 [ring did not display]
[image]

2-3-2 [ring did not display]
[image]

2-4-2[ring displayed running rainbow]
[image]

2-6-2 [ring displayed running rainbow]
[image]

3-4-3 [ring displayed running rainbow]
[image]

4-2-4 [ring displayed running rainbow]
[image]

1-10-1 [ring displayed running rainbow]
[image]

1-12-1 [ring displayed running rainbow]
[image]

1-14-1 [ring displayed running rainbow]
[image]

1-16-1 [ring displayed running rainbow]
[image]

1-17-1 [did not compile, error _PIOASMError: delay too large_]

lurch, 2021-02-24:
Thanks @psitech 

So combining your 3-4-3 timings with my table above (and calculating PIO's value's as T0H = T1, T0L = T2 + T3, T1H = T1 + T2, T1L = T3), and a PIO frequency of 8MHz, I guess that gives us:

|  | WS2812 | WS2812B | SK6812 | 3-4-3 timing | original 2-5-3 timing |
| --- | --- | --- | --- | --- | --- |
| T0H | 0.35us +/- 0.15us | 0.4us +/- 0.15us | 0.3us +/- 0.15us | 0.375us | 0.25us |
| T0L | 0.8us +/- 0.15us | 0.85us +/- 0.15us | 0.9us +/- 0.15us | 0.875us | 1.0us |
| T1H | 0.7us +/- 0.15us | 0.8us +/- 0.15us | 0.6us +/- 0.15us | 0.875us | 0.875us |
| T1L | 0.6us +/- 0.15us | 0.45us +/- 0.15us | 0.6us +/- 0.15us | 0.375us | 0.375us |

(Assuming I've got my maths correct, of course! :laughing: )

psitech, 2021-02-24:
Hi Andrew, your extended table is correct.

psitech, 2021-02-24:
If I'm correct, a 3-3-4 timing fits all 3 LEDs:
T0H = 0.375us
T0L = 0.875us
T1H = 0.750us
T1L = 0.500us

lurch, 2021-02-24:
@ladyada @ptorrone @tannewt You probably have access to more different types of NeoPixel than anybody else :grin: 
Would you mind trying the T1=3 T2=3 T3=4 timings suggested above (for https://github.com/raspberrypi/pico-examples/tree/master/pio/ws2812 ) on a selection of different types, to see if they all work reliably?

psitech, 2021-02-24:
For completeness, here is the logic analyzer output for the 3-3-4 timing:

[image]
[image]

tannewt, 2021-02-24:
> @ladyada @ptorrone @tannewt You probably have access to more different types of NeoPixel than anybody else grin
> Would you mind trying the T1=3 T2=3 T3=4 timings suggested above (for [...]

So I hit a similar question when doing Neopixel for CircuitPython. All of the datasheets have different values. I asked @paintyourdragon, who is our resident neopixel expert, and he suggested to simply think of it as a 800khz signal with 1/3 duty cycle for zero and 2/3 for one.

I [implemented it in circuitpython](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/neopixel_write/__init__.c) with:

```
bitloop:
  out x 1        side 0 [1]; Side-set still takes place before instruction stalls
  jmp !x do_zero side 1 [1]; Branch on the bit we shifted out after 1/3 duty delay. Positive pulse
do_one:
  jmp  bitloop   side 1 [1]; Continue driving high, for a long pulse
do_zero:
  nop            side 0 [1]; Or drive low, for a short pulse
```

So its 2-2-2 timing and we clock the state machine at 800khz * 6.

I haven't heard any complaints about our implementation's timing so I assume it's pretty good. We've had two bugs but neither were timing related. One was that we deinit too early and therefore didn't transmit all the bits and the other was due to us floating the pin while not transmitting.

psitech, 2021-02-24:
Hi Scott, 
I implemented your 1/3 & 2/3 dutycycle in MicroPython and that worked fine on the Neopixel ring as well. Had to change the statemachine clock to 4.8MHz.
Here is the code:
```
@rp2.asm_pio(sideset_init=rp2.PIO.OUT_LOW, out_shiftdir=rp2.PIO.SHIFT_LEFT, autopull=True, pull_thresh=24)
def ws2812():
    T1 = 2
    T2 = 2
    T3 = 2
    wrap_target()
    label("bitloop")
    out(x, 1)               .side(0)    [T3 - 1]
    jmp(not_x, "do_zero")   .side(1)    [T1 - 1]
    jmp("bitloop")          .side(1)    [T2 - 1]
    label("do_zero")
    nop()                   .side(0)    [T2 - 1]
    wrap()

# Create the StateMachine with the ws2812 program, outputting on pin
sm = rp2.StateMachine(0, ws2812, freq=4_800_000, sideset_base=Pin(PIN_NUM))

# Start the StateMachine, it will wait for data on its FIFO.
sm.active(1)
```
Here is the logic analyzer output:
[image]
[image]

One remark: this 2-2-2 timing theoretically violates the T1H & T1L spec of the SK6812...

Regards,
Paul

tannewt, 2021-02-24:
> One remark: this 2-2-2 timing theoretically violates the T1H & T1L spec of the SK6812...

I believe most modern NeoPixels we sell are the SK6812 so I wouldn't worry about it. Many products say they may be WS2812B or SK6812 and used interchangeably. I bet SK6812 are one of the best tested.

lurch, 2021-02-24:
> One remark: this 2-2-2 timing theoretically violates the T1H & T1L spec of the SK6812...

I'm not entirely convinced by the timing values in the SK6812 datasheet, since the maximum T0H (0.3us + 0.15us) is the same as the minimum T1H (0.6us - 0.15us), and the minimum T0L (0.9us - 0.15us) is the same as the maximum T1L (0.6us + 0.15us) ? :thinking: :confused:

lurch, 2021-02-25:
I'm not one of the chip designers, but personally I'd be tempted to go with the 2-2-2 @ 4.8MHz that @tannewt has confirmed works well for CircuitPython. Hard to argue against real-world testing! :wink:

lurch, 2021-02-25:
Ping also @benevpi since I believe he's also done quite a bit of stuff with NeoPixels on the Raspberry Pi Pico.

psitech, 2021-02-26:
I'm fine with the 2-2-2 @ 4.8MHz timing.

benevpi, 2021-03-02:
I've only just seen this as it seems my spam filter's decided it doesn't like Github. I'll dig out my big pile of test WS2812B-alikes and test them out with the 2-2-2 timings.