raspberrypi/pico-examples #54: uart_advanced.c appears to incorporate a 1 second delay

MrYsLab, 2021-02-16:
I have documented the issue on [the raspberry pi forum.](https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=304169&p=1821187#p1821187)
In my example on the forum, I slightly modified the example to match my Python client code (also on the forum post), but if I run uart_advanced.c unmodified I see similar behavior.

lurch, 2021-02-18:
There's too many "moving parts" in your forum post to be able to tell what's going on.
Please try to produce a standalone reproducible example here in this GH issue (without referencing forum posts), using the serial UART on a Raspberry Pi, as shown in section 4.5 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

MrYsLab, 2021-02-18:
@lurch As mentioned above, if I run the stock uart_advanced.c  driving it with my Python test script I see the delay. If I run the same Python script using a modified version of hello_usb.c (to loop; the data back), there is no delay. 

The only variable is the C application. I still believe there is an issue. Have you run the example and if you are not seeing any delay, what are you using as a data driver?