raspberrypi/pico-examples #56: CMake fails to build examples at the command line on Windows

slrAU, 2021-02-18:
Hi, 

Just a heads up.  I found that the pico examples weren't building and I needed to add:

> set(PICO_SDK_PATH $ENV{PICO_SDK_PATH})

... to the start of the CMakeLists.txt file. Just before where it says "include(pico_sdk_import.cmake)".

Without this line, CMake was not using the PICO_SDK_PATH environment variable, as per section 8.2.3 of the Getting Started with Raspberry Pi Pico document.

Cheers.

lurch, 2021-02-18:
Hmmm, I wonder why https://github.com/raspberrypi/pico-examples/blob/master/pico_sdk_import.cmake#L6 isn't working...  Any ideas @kilograham ?

2bndy5, 2021-04-01:
+1 to this. I was hitting the same problem (on windows 10 following _getting-started-with-pico_), and @slrAU solution fixed it. Thanks @slrAU !

FYI I did a reboot after confirming the env var was properly set (as instructed), but the env var was being used. Full error ouput if it helps:
```text
PS C:\Users\ytreh\Documents\GitHub\pico-examples\build> cmake -G "NMake Makefiles" ..
CMake Error at pico_sdk_import.cmake:52 (message):
  Directory 'C:/Users/ytreh/Documents/GitHub/pico-examples/build/pico-sdk'
  not found
Call Stack (most recent call first):
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
```

For passers-by: OP says 
> section 8.2.3 of the Getting Started with Raspberry Pi Pico document

but it is actually from section 9.2.3

lurch, 2021-04-01:
@2bndy5 Thanks. Just to clarify, are you saying that it failed when you had `PICO_SDK_PATH` set in the "USER env vars", but worked when you had  `PICO_SDK_PATH` set in the "SYSTEM env vars" ? I assume you were setting the same path both times? :wink: 

Does the "PS" in your output above indicate that you're using PowerShell? Our [getting started](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) instructions say to use a Developer Command Prompt.

2bndy5, 2021-04-01:
>are you saying that it failed when you had PICO_SDK_PATH set in the "USER env vars", but worked when you had PICO_SDK_PATH set in the "SYSTEM env vars" ? 

That's correct

>I assume you were setting the same path both times?

 yes. [as instructed](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf#%5B%7B%22num%22%3A45%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C672.045%2Cnull%5D) I set it o `..\..\pico-sdk` using `setx PICO_SDK_PATH "..\..\pico-sdk"` cmd from a developer prompt. At first I just restarted the prompt so it would take affect, then I restarted my PC; still no joy until I moved the env var into SYSTEM env vars.

Just to be very clear:
1. I deleted the fix from @slrAU in the _CMakeLists.txt_
2. I removed the PICO_SDK_PATH from the USERS env vars  (using the windows dialogue). I can confirm that the `setx PICO_SDK_PATH "..\..\pico-sdk"` did create the env var (but in USER space)
3. I added the PICO_SDK_PATH to the SYSTEM env vars (using the windows dialogue)
4. I restarted my PC then booted into up the developer prompt again.

I'm now currently (& successfully) using VSCode to build/compile following the rest of the getting-started guide (with a Feather RP2040).

lurch, 2021-04-01:
Are you also launching VSCode from the Developer Command Prompt?

2bndy5, 2021-04-01:
I was. but I was using `code .` to skip having to File->Open Folder that the developer prompt was already in.

jksevend, 2022-04-16:
@slrAU Wow.

After one and a half days of trying and stumbling upon your issue i was finally relieved.
It was so confusing because the pico-examples were all building but my own "custom" blink was not building. 

I tried having the pico_sdk_init.cmake inside my project and just including that like in the examples but in the end only your solution worked out for me. (Windows 11 Following this tutorial https://www.arnabkumardas.com/platforms/rp2040/raspberry-pi-pico-sdk-and-toolchain-setup-in-windows-10/)

closed 2022-05-13 by kilograham

kjrandez, 2022-10-24:
Can't fathom why this was closed. The instructions must be wrong or incomplete.
Why would cmake not be able to use a User level env var?
Anyways I just pass -DPICO_SDK_PATH="..\..\pico-sdk" when calling CMake.
It's also really misguided that the instructions would suggest using a relative path for the whole system to share

geraldombuthia, 2023-08-12:
> +1 to this. I was hitting the same problem (on windows 10 following _getting-started-with-pico_), and @slrAU solution fixed it. Thanks @slrAU !
> 
> FYI I did a reboot after confirming the env var was properly set (as instructed), but the env var was not being used. Full error ouput if it helps:
> 
> ```
> PS C:\Users\ytreh\Documents\GitHub\pico-examples\build> cmake -G "NMake Makefiles" ..
> CMake Error at pico_sdk_import.cmake:52 (message):
>   Directory 'C:/Users/ytreh/Documents/GitHub/pico-examples/build/pico-sdk'
>   not found
> Call Stack (most recent call first):
>   CMakeLists.txt:4 (include)
> 
> 
> -- Configuring incomplete, errors occurred!
> ```
> 
> For passers-by: OP says
> 
> > section 8.2.3 of the Getting Started with Raspberry Pi Pico document
> 
> but it is actually from section 9.2.3
> 
> ## UPDATE
> @lurch @kilograham I removed the fix from @slrAU and moved the `PICO_SDK_PATH` env var from the USER env vars to the SYSTEM env vars. Then I rebooted, deleted then re-created the build folder, and ran the `cmake -G "NMake Makefiles" ..` cmd from the build folder; it worked no problems (no modification of _CMakeLists.txt_ needed).

This worked for me. It appears once I opened a particular work file the Path was being set to ..\..\path which was leading nowhere.