raspberrypi/pico-examples #63: Add a USB "dev_lowlevel" device example but using tinyusb

profi200, 2021-02-20:
There is a "dev_lowlevel" example but it's a little too low level for my taste. Unless i'm blind the tinyusb README states it does support generic devices with simple in and out endpoints ("- Vendor-specific class support with generic In & Out endpoints. Can be used with MS OS 2.0 compatible descriptor to load winUSB driver without INF file.") but there is no example provided (both by this repo and tinyusb itself).

Would be nice if an example could be added that does the same as "dev_lowlevel" but using tinyusb. Thanks.

profi200, 2021-02-24:
Update:
I have modified the dev_hid_generic_inout example so it works exactly as i want.

2 things i noticed:
- Endpoint buffer size seems to be restricted to 64 bytes which means lots of IRQs :/
- TinyUSB doesn't recognize there is incoming data anymore if the PC side test code connects again causing it to not respond. The workaround is resetting the USB device. Unsure if this is expected behavior or a bug in TinyUSB/the underlying USB driver.

Here is the code:
tusb_config.h
```
/*
 * The MIT License (MIT)
 *
 * Copyright (c) 2019 Ha Thach (tinyusb.org)
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 *
 */

#ifndef _TUSB_CONFIG_H_
#define _TUSB_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

//--------------------------------------------------------------------
// COMMON CONFIGURATION
//--------------------------------------------------------------------

// defined by compiler flags for flexibility
#ifndef CFG_TUSB_MCU
#error CFG_TUSB_MCU must be defined
#endif

#if CFG_TUSB_MCU == OPT_MCU_LPC18XX || CFG_TUSB_MCU == OPT_MCU_LPC43XX || CFG_TUSB_MCU == OPT_MCU_MIMXRT10XX || \
    CFG_TUSB_MCU == OPT_MCU_NUC505 || CFG_TUSB_MCU == OPT_MCU_CXD56
#define CFG_TUSB_RHPORT0_MODE     (OPT_MODE_DEVICE | OPT_MODE_HIGH_SPEED)
#else
#define CFG_TUSB_RHPORT0_MODE     OPT_MODE_DEVICE
#endif

#ifndef CFG_TUSB_OS
#define CFG_TUSB_OS                 OPT_OS_PICO
#endif

// CFG_TUSB_DEBUG is defined by compiler in DEBUG build
// #define CFG_TUSB_DEBUG           0

/* USB DMA on some MCUs can only access a specific SRAM region with restriction on alignment.
 * Tinyusb use follows macros to declare transferring memory so that they can be put
 * into those specific section.
 * e.g
 * - CFG_TUSB_MEM SECTION : __attribute__ (( section(".usb_ram") ))
[... 412 lines cut ...]
	if(libusb_reset_device(devH))
	{
		fprintf(stderr, "Failed to reset device.\n");
		libusb_close(devH);
		libusb_exit(NULL);
		return 3;
	}

	if(libusb_claim_interface(devH, 0))
	{
		fprintf(stderr, "Failed to claim interface 0.\n");
		libusb_close(devH);
		libusb_exit(NULL);
		return 4;
	}

	sendReceiveHello(devH);

	libusb_release_interface(devH, 0);
	libusb_close(devH);
	libusb_exit(NULL);

	return 0;
}
```

lurch, 2021-02-24:
ping @liamfraser

liamfraser, 2021-02-24:
64 byte is the max packet size of a USB 1.1 device. That's just how the protocol works. https://github.com/raspberrypi/picoprobe is an example of a vendor device.

liamfraser, 2021-02-24:
I don't have to reset the device before I can talk to it in picoprobe but I am getting the data differently:

```
    if ( tud_vendor_available() ) {
        uint count = tud_vendor_read(&probe.rx_buf[probe.rx_len], 64);
```

It depends when tinyusb decides to call the callback. It might only do it when you get a full 64 bytes?

profi200, 2021-02-25:
Yeah, almost expected that to be a USB 1.1 limitation. That's a shame because as mentioned that will cause lots of IRQs for larger transfers.

The problem with having to reset the device i dunno. The PC side test code is actually a stripped down version of code i have written for something else long ago and it didn't need a reset either (but the device acted buggy in other ways like timing out on first try). It doesn't have issues with arbitrary sized bulk transfers if the test code opens the device for the first time. Can send anything up to 64 bytes and multiple bulk transfers as well.

libusb output without reset first try:
```
[blub@blub-PC vendor_generic_inout]$ sudo ./pico_usb_test
[sudo] Passwort für blub: 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.003590] [000297a0] libusb: debug [libusb_get_device_list]  
[ 0.003608] [000297a0] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.003614] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003621] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003625] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003627] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003629] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003631] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003634] [000297a0] libusb: debug [libusb_get_device_descriptor]  
[ 0.003638] [000297a0] libusb: debug [libusb_open] open 2.7
[ 0.112770] [000297a0] libusb: debug [usbi_add_event_source] add fd 7 events 4
[ 0.112825] [000297a0] libusb: debug [libusb_set_configuration] configuration 1
[ 0.115049] [000297a0] libusb: debug [libusb_claim_interface] interface 0
[ 0.115139] [000297a0] libusb: debug [libusb_alloc_transfer] transfer 0x555e464e0f78
[ 0.115155] [000297a0] libusb: debug [libusb_submit_transfer] transfer 0x555e464e0f78
[ 0.115166] [000297a0] libusb: debug [add_to_flying_list] arm timer for timeout in 5000ms (first in line)
[ 0.115184] [000297a0] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 11
[ 0.115218] [000297a0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.115230] [000297a0] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.115244] [000297a0] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.115265] [000297a0] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.115279] [000297a0] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=11
[ 0.115291] [000297a0] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.115299] [000297a0] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[ 0.115307] [000297a0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.115323] [000297a0] libusb: debug [usbi_handle_transfer_completion] transfer 0x555e464e0f78 has callback 0x7f85889837d0
[ 0.115335] [000297a0] libusb: debug [sync_transfer_cb] actual_length=11
[ 0.115350] [000297a0] libusb: debug [libusb_free_transfer] transfer 0x555e464e0f78
[ 0.115363] [000297a0] libusb: debug [libusb_alloc_transfer] transfer 0x555e464dabb8
[ 0.115371] [000297a0] libusb: debug [libusb_submit_transfer] transfer 0x555e464dabb8
[ 0.115380] [000297a0] libusb: debug [add_to_flying_list] arm timer for timeout in 5000ms (first in line)
[ 0.115396] [000297a0] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 11
[ 0.115417] [000297a0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.115428] [000297a0] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.115465] [000297a0] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.115481] [000297a0] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=11
[ 0.115492] [000297a0] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.115504] [000297a0] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
[ 0.115515] [000297a0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.115529] [000297a0] libusb: debug [usbi_handle_transfer_completion] transfer 0x555e464dabb8 has callback 0x7f85889837d0
[ 0.115541] [000297a0] libusb: debug [sync_transfer_cb] actual_length=11
[ 0.115556] [000297a0] libusb: debug [libusb_free_transfer] transfer 0x555e464dabb8
Received: "Hello USB!"
[ 0.115598] [000297a0] libusb: debug [libusb_release_interface] interface 0
[ 0.115637] [000297a0] libusb: debug [libusb_close]  
[ 0.115651] [000297a0] libusb: debug [usbi_remove_event_source] remove fd 7
[ 0.115677] [000297a0] libusb: debug [libusb_exit]  
[ 0.115689] [000297a0] libusb: debug [libusb_exit] destroying default context
[ 0.115700] [000297a0] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.115711] [000297a0] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.115727] [000297a0] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[ 0.115785] [000297a0] libusb: debug [usbi_wait_for_events] poll() returned 0
[ 0.115801] [000297a0] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.115816] [000297a0] libusb: debug [libusb_unref_device] destroy device 3.1
[... cut ...]
[ 5.114984] [000297a9] libusb: debug [libusb_unref_device] destroy device 2.2
[ 5.115003] [000297a9] libusb: debug [libusb_unref_device] destroy device 2.5
[ 5.115021] [000297a9] libusb: debug [libusb_unref_device] destroy device 2.6
[ 5.115039] [000297a9] libusb: debug [libusb_unref_device] destroy device 2.1
[ 5.115056] [000297a9] libusb: debug [usbi_remove_event_source] remove fd 6
[ 5.115088] [000297a9] libusb: debug [usbi_remove_event_source] remove fd 5
[ 5.115154] [000297aa] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

JesperNaarttijarvi, 2024-06-05:
Sorry for the very late response, but did you get it to work? 

I wanted to achieve the same test result, and got stuck. Sending data works, but not receiving it. 

Tried with your code, but also gets stuck on "Failed to receive data."

My logs are very similar to yours:
HOST
```
sudo ./cm4_usb_host 1

Starting USB scan thread...
2024-06-05 10:43:00 Found a Pico USB device
2024-06-05 10:43:00 Setting up communications
2024-06-05 10:43:00 Sending counter 200 to device
Sent 1 bytes
2024-06-05 10:43:00 Waiting for response from device...
Error receiving data from device: LIBUSB_ERROR_TIMEOUT
2024-06-05 10:43:01 Sending counter 201 to device
Sent 1 bytes
2024-06-05 10:43:01 Waiting for response from device...
Error receiving data from device: LIBUSB_ERROR_TIMEOUT
2024-06-05 10:43:03 Sending counter 202 to device
Sent 1 bytes
2024-06-05 10:43:03 Waiting for response from device...
Error receiving data from device: LIBUSB_ERROR_TIMEOUT
2024-06-05 10:43:04 Sending counter 203 to device
Sent 1 bytes
2024-06-05 10:43:04 Waiting for response from device...
Error receiving data from device: LIBUSB_ERROR_TIMEOUT
```

DEVICE (RP2040 OVER UART CONSOLE):
```
Sending counter 201 to host, 1 bytes written
USBD Xfer Complete on EP 01 with 1 bytes
  VENDOR xfer callback
  Queue EP 01 with 64 bytes ...
Received counter 201 from host
Sending counter 202 to host, 1 bytes written
USBD Xfer Complete on EP 01 with 1 bytes
  VENDOR xfer callback
  Queue EP 01 with 64 bytes ...
Received counter 202 from host
Sending counter 203 to host, 1 bytes written
USBD Xfer Complete on EP 01 with 1 bytes
  VENDOR xfer callback
  Queue EP 01 with 64 bytes ...
Received counter 203 from host
Sending counter 204 to host, 1 bytes written
```