raspberrypi/pico-examples #86: Wrong condition in picoboard/blinky

hotovson, 2021-03-10:
there is wrong condition on lines
https://github.com/raspberrypi/pico-examples/blob/2fb08a028f886624ebc9c938775e054860549159/picoboard/blinky/blinky.c#L57
and
https://github.com/raspberrypi/pico-examples/blob/2fb08a028f886624ebc9c938775e054860549159/picoboard/blinky/blinky.c#L59

must be 
`if (*str >= 'A' && *str <= 'Z') {` 
and 
`} else if (*str >= 'a' && *str <= 'z') {`
respectively

lurch, 2021-03-10:
Hah, just took me about 10 minutes to see what the difference was!! :laughing: :mag: 
In the version at https://github.com/raspberrypi/pico-examples/blob/develop/picoboard/blinky/blinky.c it's using `< 'z'` but it should be `<= 'z'` - which we obviously didn't notice ourselves because the example `"Hello world"` string doesn't contain a `z`.
Thanks for spotting this, I'll submit a PR...

lurch, 2021-04-13:
This can be closed now @kilograham

lurch, 2021-04-13:
@kilograham Ahhhh, so [this](https://docs.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue) is why issues aren't being closed when the associated PR is merged:
> Note: The special keywords in a pull request description are interpreted when the pull request targets the repository's default branch. However, if the PR's base is any other branch, then these keywords are ignored, no links are created and merging the PR has no effect on the issues.

Looks like there's a 3rd-party "action" to get around that, if this behaviour becomes too annoying https://github.com/marketplace/actions/closes-issues-related-to-a-merged-pull-request
(but I dunno the security implications of giving a 3rd-party addon access to `secrets.GITHUB_TOKEN` )

PR #87 (2021-04-13): Fix condition for printing morse-code Z
Fixes #86
commit 612469d3 (2021-04-13): Fix condition for printing morse-code Z

Fixes #86
closed 2021-06-03 by kilograham