raspberrypi/pico-examples #88: Incorrect blink frequency in pio_blink

sparkyhall, 2021-03-14:
The formula used in the recent update to use the sys clock is incorrect.

The offending line is the last line in blink.c:

`pio->txf[sm] = clock_get_hz(clk_sys) / 2 * freq;`

This divides sys clock by 2 and multiplies the result by the frequency desired. 

The result is that a frequency of 1 Hz correctly blinks at 1Hz, however, a frequency of 4Hz blinks at 0.25Hz and a frequency of 3Hz at 0.33Hz.

Replacing the multiply with divide corrects this.

rschaer, 2021-04-14:
Just noticed that too.` pio->txf[sm] = (clock_get_hz(clk_sys)/2) / freq; ` works fine.

Reneg973, 2021-05-01:
Or - if you include rounding:
`pio->txf[sm] = (clock_get_hz(clk_sys) + freq) / (2 * freq);`

lurch, 2021-07-12:
Fixed by #121

kilograham, 2022-05-13:
merged into develop (a long time ago)

closed 2022-05-13 by kilograham