raspberrypi/pico-examples #89: /dev/ttyACM0 missing when running I2C examples

JustinS-B, 2021-03-17:
I'm running Arch linux, I have installed the **SDK** then cloned **pico-examples**, **pico-playground** and **pico-extras** and added them to PATH in my .bashrc

I have also added **export PICO_SDK_PATH=/usr/share/pico-sdk** to my .bashrc

I create the build directory, cd into it, **cmake ..** then **make** and the examples compile

**flash_program.uf2** compiles without error, and it runs when I drop it into my Pico, and the LED flashes as expected

**hello_usb.uf2** also appears to compile without error, and it also runs when I drop it into my Pico. I can see **/dev/ttyACM0** and **sudo minicom** gives me the expected "Hello, world!" when I point it at **/dev/ttyACM0**

**mpu6050_i2c.uf2**, **lcd_1602_i2c.uf2** and **i2c_bus_scan.uf2** compile with the following warnings, **/dev/ttyACM0** is not created and nothing appears to work using a known good MPU6050. I have used a different computer (running Arch again), along with a different known good MPU6050 and a different Pico - same results.

What did I miss or forget to do please?

```
<snip>

Scanning dependencies of target i2c_bus_scan
[ 55%] Building C object i2c/bus_scan/CMakeFiles/i2c_bus_scan.dir/bus_scan.c.obj
/home/user/GitHub/pico-examples/i2c/bus_scan/bus_scan.c: In function 'main':
/home/user/GitHub/pico-examples/i2c/bus_scan/bus_scan.c:38:2: warning: #warning i2c/bus_scan example requires a board with I2C pins [-Wcpp]
   38 | #warning i2c/bus_scan example requires a board with I2C pins
      |  ^~~~~~~

<snip>

[ 58%] Built target i2c_bus_scan
Scanning dependencies of target lcd_1602_i2c
[ 58%] Building C object i2c/lcd_1602_i2c/CMakeFiles/lcd_1602_i2c.dir/lcd_1602_i2c.c.obj
/home/user/GitHub/pico-examples/i2c/lcd_1602_i2c/lcd_1602_i2c.c: In function 'main':
/home/user/GitHub/pico-examples/i2c/lcd_1602_i2c/lcd_1602_i2c.c:134:6: warning: #warning i2c/lcd_1602_i2c example requires a board with I2C pins [-Wcpp]
  134 |     #warning i2c/lcd_1602_i2c example requires a board with I2C pins
      |      ^~~~~~~

<snip>

[ 61%] Built target lcd_1602_i2c
Scanning dependencies of target mpu6050_i2c
[ 61%] Building C object i2c/mpu6050_i2c/CMakeFiles/mpu6050_i2c.dir/mpu6050_i2c.c.obj
/home/user/GitHub/pico-examples/i2c/mpu6050_i2c/mpu6050_i2c.c: In function 'main':
/home/user/GitHub/pico-examples/i2c/mpu6050_i2c/mpu6050_i2c.c:83:6: warning: #warning i2c/mpu6050_i2c example requires a board with I2C pins [-Wcpp]
   83 |     #warning i2c/mpu6050_i2c example requires a board with I2C pins
      |      ^~~~~~~

<snip>
```

fivdi, 2021-03-22:
@JustinS-B Is this issue the same as the issue that was discussed and resolved on the Raspberry Pi forum at https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=307375? If so, please close this GitHub issue.

lurch, 2021-03-22:
> flash_program.uf2 compiles without error, and it runs when I drop it into my Pico, and the LED flashes as expected

That example specifically demonstrates writing to flash, the example for blinking the LED is `blink` :slightly_smiling_face: 

> hello_usb.uf2 also appears to compile without error, and it also runs when I drop it into my Pico. I can see /dev/ttyACM0
mpu6050_i2c.uf2, lcd_1602_i2c.uf2 and i2c_bus_scan.uf2 compile with the following warnings, /dev/ttyACM0 is not created

IIRC `hello_usb` is the **only** example that uses USB stdio, all the other examples (that use stdio) use UART stdio (because this makes debugging easier). It's easy to update each of the examples to use USB stdio though, see chapter 4 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf

The compile-warnings you're seeing look like you might be building `pico-examples` against an older version of `pico-sdk`? They're both updated reasonably regularly.

JustinS-B, 2021-03-23:
Thank you both for your patient replies. As @lurch pointed out, the important bit of information that I had missed was that all the examples default to UART stdio, which makes life really simple if you are running everything from a Raspberry Pi.

@lurch Yes, you are right, flash_program is not blink, sorry, I copied the wrong file name by accident.
@fivdi Yes, that Pi forum post describes how to enable USB stdio for the I2C sketches, as opposed to the default UART stdio

However, for the record, the errors that I listed above can easily be resolved by adding these definitions like these to the example that you need to compile.

```
#define i2c_default i2c0
#define PICO_DEFAULT_I2C_SDA_PIN 4
#define PICO_DEFAULT_I2C_SCL_PIN 5
```

closed 2021-03-23 by JustinS-B

lurch, 2021-03-23:
> However, for the record, the errors that I listed above can easily be resolved by adding these definitions like these to the example that you need to compile.

... which are also set in https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/pico.h which is why I was wondering if you were using an older pico-sdk (from before these were added) ?

JustinS-B, 2021-03-23:
@lurch yes, sorry, you are 100% right on that one as well. I had used an Arch Linux AUR installer for the SDK, which it turns out wasn't updating. As soon as I  uninstalled it and cloned the SDK myself, all of those compile errors instantly stopped. Thank you.