raspberrypi/pico-examples #90: hello_serial.c produces garbled output in minicom if Bluetooth is enabled

mslinklater, 2021-03-18:
Documentation probably needs amended to say that Bluetooth needs to be disabled for hello_serial.c to work properly.

See this thread:

https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=305731&p=1838457#p1838457

lurch, 2021-03-18:
Did you follow section 4.5 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf ?

mslinklater, 2021-03-18:
Yes, but I had Bluetooth enabled and the output via minicom was garbled.

When I added 'dtoverlay=disable-bt' to my config.txt it worked perfectly.

lurch, 2021-03-18:
My understanding was that using the `raspi-config` option documented in the Pico "Getting Started" guide should always give you a "usable" UART connection? Or do we need to add extra steps if the user also has bluetooth enabled @XECDesign @pelwell ?

@aallan Could you move this to the pico-feedback repo please?

aallan, 2021-03-18:
The serial console on GPIO and Bluetooth serial were at one point using the same hooks. I vaguely remember this changing sometime around the Pi 3. Don't know what the current state of things is though?

lurch, 2021-03-18:
https://github.com/raspberrypi/documentation/blob/master/configuration/uart.md :wink: 

But like I said, I thought `raspi-config` automatically did all the necessary steps?