raspberrypi/pico-examples #93: dev_hid_generic_inout Appears to hang when reading

TaraHoleInIt, 2021-03-21:
Hi.
I've tried using hid_test.py from the tinyusb examples to test this example, but it seems to hang the Pico when trying to read.

Host system:
Macbook Pro M1
MacOS Big Sur 11.2.3
Connected via USB hub

nachiketathakur, 2022-03-16:
I am having the same issue. Could you solve it? I have observed that when you flash the *.uf2 file present in pico-examples, it works just fine.

lurch, 2022-03-17:
Huh? https://github.com/raspberrypi/pico-examples/search?q=dev_hid_generic_inout&type=code
:confused:

nachiketathakur, 2022-03-17:
> Huh? https://github.com/raspberrypi/pico-examples/search?q=dev_hid_generic_inout&type=code 😕

The link is broken. It says "We couldn’t find any code matching 'dev_hid_generic_inout' in [raspberrypi/pico-examples]"

lurch, 2022-03-17:
Yeah, exactly. Could you explain where the `dev_hid_generic_inout` you're talking about is located, because according to GitHub it's not in this repo? I'm also unclear what you mean by "I have observed that when you flash the *.uf2 file present in pico-examples, it works just fine." ?

nachiketathakur, 2022-03-17:
I should have been more clear.
When you build the whole example directory, in the build directory of pico examples, the *.UF2 file of this particular example as well as other examples is built, as mentioned on the pico-examples github page in USB Device section: 

> Most of the USB device examples come directly from the TinyUSB device examples directory [here](https://github.com/hathach/tinyusb/tree/master/examples/device). Those that are supported on RP2040 devices are automatically included as part of the pico-examples build as targets named tinyusb_dev_<example_name> e.g. https://github.com/hathach/tinyusb/tree/master/examples/device/hid_composite is built as tinyusb_dev_hid_composite

I flashed the tinyusb_dev_hid_generic_inout.uf2 file located at pico-examples\build\usb\device\tinyusb_device_examples\hid_generic_inout\ and it worked as expected with the python script that is provided with the example on TinyUSB's github page.

lurch, 2022-03-17:
Ahh! Thanks for the extra explanation :+1: 
tinyusb_dev_hid_generic_inout comes from the tinyusb repo, so that's indeed why I couldn't find it in `pico-examples`.

> it worked as expected

That's great to hear!

So has this problem been solved, or could you explain what you meant with "I am having the same issue." ?

nachiketathakur, 2022-03-17:
The problem is not solved yet. 

When I copy the source code of the example from tinyUSB and keep other files like tusb_config.h, usb_descriptors.c and usb_descriptors.h 'as it is' as they are in the example code of dev_hid_composite(which also comes with pico-examples), I am having the issue where the python script will take the input, but it just hangs there. It won't loop back the sent string. 

But the python script works fine when I simply flash the *.uf2 file which is built from TinyUSB source. 

I thought there must be some USB driver issues (just a guess), so I tried it on Linux and the problem remained. 

Also, when I copy paste the tusb_config.h file that comes with TinyUSB examples, I get errors in the macros.

lurch, 2022-03-17:
I'm not sure I follow, so pinging @hathach as he probably has a better understanding of what's going on.

TaraHoleInIt, 2022-03-17:
I wish I had written more about this in-depth when I first posted about it, because now I remember nothing about the circumstances around it.
I'll see if I can once again reproduce it on my end.