raspberrypi/pico-examples #94: Example request RS485 rx

CharlesGodwin, 2021-03-21:
I have many skills but I am completely ignorant about hardware interfaces. 
I need to read a RS485 data stream. It's sort of a firehose data stream DATA, Pause, DATA... and I currently use a Pi and a RS485 USB dongle for the read.  I sample periodically as I can't just sit on the feed and read every byte. I believe, if I could read the stream, I could move all my specialized management of the data to a Pico. I just need to be able to get those bytes and inter record gaps. ;)

This is not an urgent request, just a nice to have feature.

lurch, 2021-03-21:
Sounds like another job for the wonderful [PIO](https://www.raspberrypi.org/blog/what-is-pio/) - but I _suspect_ that none of the RPi engineers will ever get time to look at implementing this themselves so you'd have to wait for a community-member to implement it.

Hmm, although https://en.wikipedia.org/wiki/RS-485#Standard says "RS-485 only specifies the electrical characteristics of the generator and the receiver: the physical layer. It does not specify or recommend any communications protocol; Other standards define the protocols for communication over an RS-485 link." and it sounds like your "RS485 USB dongle" is currently handling the physical layer, and it's actually the communications protocol ("firehose data stream") that you're needing help with?

> I sample periodically as I can't just sit on the feed and read every byte.

Perhaps you need to look into multi-threaded programming and/or the use of interrupts? :slightly_smiling_face:

CharlesGodwin, 2021-03-21:
I was looking at PIO as the solution. When the Pico was announced, someone suggested in the comments they may have done a RS485 implementation as a proof of concept. But nothing more was mentioned.

The dongle works for now but it's a packaging issue. I'd really like to create a purpose built USB device that had a straightforward serial interface. My current implementation is [here](https://pypi.org/project/pymagnum/).

I have experience with multithreading / multiprocessors, it's the hardware serial read I'm ignorant about. I plan one core to do the RS485 read and one core to do the formatting and talking to the outside world.

This is a "nice to have" project so I'll wait.

kripton, 2021-12-15:
As @lurch mentioned, RS-485 is only the electrical specification, not the content sent over it. So since the voltages used for RS-485 are higher than 5V, you should not connect your Pico directly to the bus (it will probably kill the chip or at least some pins of it). You will need some kind of RS-485-receiver in any case (the RS-485-USB-Dongle you mentioned is probably just that plus some UART-to-USB-Interface). Examples for RS-485-transceivers are the ICs "MAX485", "SN75176", "ADM3493" (all not galvanically isolated) or the "ADM2483"/"ADM2486". For most of them, there are cheap break-out-boards available so you can connect them easily to your Pico-board.

If you have the hardware set-up, reading the data stream can easily be done using one of the RP2040's UART- or PIO-peripherals. Depending on the data you expect (Modbus, DMX, ...?), you could use the PIO UART RX example: https://github.com/raspberrypi/pico-examples/tree/master/pio/uart_rx. If it's DMX (or something different, just another example), this might help you: https://github.com/jostlowe/Pico-DMX/blob/main/extras/DmxInput.pio