raspberrypi/pico-examples #100: /dev/ttyACM0 not showing up

Florisvdf, 2021-04-10:
Hi,

I'm trying to run the examples from the Getting Started with Raspberry Pi Pico pdf. I'm on Big Sur 11.2.3 with the Apple M1 chip. After downloading `cmake` and `arm-none-eabi-gcc`, I do the following:

```
(base) floris@Floriss-MBP pico % git clone -b master https://github.com/raspberrypi/pico-sdk.git
Cloning into 'pico-sdk'...
remote: Enumerating objects: 251, done.
remote: Counting objects: 100% (251/251), done.
remote: Compressing objects: 100% (162/162), done.
remote: Total 2721 (delta 107), reused 131 (delta 57), pack-reused 2470
Receiving objects: 100% (2721/2721), 1.47 MiB | 4.92 MiB/s, done.
Resolving deltas: 100% (1214/1214), done.
(base) floris@Floriss-MBP pico % cd pico-sdk
(base) floris@Floriss-MBP pico-sdk % git submodule update --init
Submodule 'tinyusb' (https://github.com/raspberrypi/tinyusb.git) registered for path 'lib/tinyusb'
Cloning into '/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb'...
Submodule path 'lib/tinyusb': checked out '11c23f88bf42f64ce14b8a7b0b2a4e207dc4dd12'
(base) floris@Floriss-MBP pico-sdk % cd ..
(base) floris@Floriss-MBP pico % git clone -b master https://github.com/raspberrypi/pico-examples.git
Cloning into 'pico-examples'...
remote: Enumerating objects: 106, done.
remote: Counting objects: 100% (106/106), done.
remote: Compressing objects: 100% (80/80), done.
remote: Total 667 (delta 44), reused 50 (delta 22), pack-reused 561
Receiving objects: 100% (667/667), 2.24 MiB | 6.82 MiB/s, done.
Resolving deltas: 100% (222/222), done.
```

I then set the `PICO_SDK_PATH` , create a `build` directory in the `pico-examples` directory, move to it and run `cmake`

```
(base) floris@Floriss-MBP pico % cd pico-examples 
(base) floris@Floriss-MBP pico-examples % mkdir build
(base) floris@Floriss-MBP pico-examples % cd build 
(base) floris@Floriss-MBP build % cmake ..
Using PICO_SDK_PATH from environment ('/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk')
PICO_SDK_PATH is /Users/floris/personal_projects/raspberry_pi/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
PICO_GCC_TRIPLE defaulted to arm-none-eabi
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
Defaulting PICO target board to pico since not specified.
Using board configuration from /Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 (found version "3.9.2") found components: Interpreter 
TinyUSB available at /Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.9.1") found components: doxygen missing components: dot
ELF2UF2 will need to be built
PIOASM will need to be built
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/floris/personal_projects/raspberry_pi/pico/pico-examples/build
```

After successfully going through the `blink` example, I move to `build/hello_world/usb` and run `make`

```
(base) floris@Floriss-MBP build % cd hello_world/usb 
(base) floris@Floriss-MBP usb % make
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is AppleClang 12.0.0.12000032
[... 75 lines cut ...]
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/device/usbd.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/device/usbd_control.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/audio/audio_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/dfu/dfu_rt_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/midi/midi_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/net/net_device.c.obj
[ 50%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/usbtmc/usbtmc_device.c.obj
[100%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/class/vendor/vendor_device.c.obj
[100%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/tusb.c.obj
[100%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj
[100%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/Users/floris/personal_projects/raspberry_pi/pico/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[100%] Linking CXX executable hello_usb.elf
[100%] Built target hello_usb
```

When I plug in the pico in the USB port while holding the BOOTSEL button and move `hello_usb.uf2` to `/Volumes/RPI-RP2/`, the pico unmounts, but `/dev/ttyACM0` does not show up. Running `sudo dmesg|grep tty` also doesn't show anything. Any idea what is causing this and what I could try to solve this? Thanks!

Cheers

lurch, 2021-04-10:
I don't have a Mac myself, but I believe that `/dev/ttyACM0` is only the device-name on Linux. According to https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf the device-name would be something like `/dev/tty.usbmodem0000000000001` on MacOS.

@aallan I guess this needs clarifying in the getting-started guide? Could you add something suitable to section 9.1.4. please?

kilograham, 2021-04-10:
Yup; i was going to suggest you try blinky first; I was wondering if there was anything weird about an M1 hosted build, but there shouldn't be.

aallan, 2021-04-10:
On a Mac the serial port name will depend on what FTDI serial dongle you use, it'll likely be something like /dev/tty.usbserial-*, e.g.

$ ls -l /dev/tty* | grep usb
crw-rw-rw- 1 root wheel 11, 66 4 Jan 14:15 /dev/tty.usbserial-A9048F6Z
$

aallan, 2021-04-10:
> Yup; i was going to suggest you try blinky first; I was wondering if there was anything weird about an M1 hosted build, but there shouldn't be.

I've had anecdotal reports that things build and work fine on M1 Macs. I'm presuming in Rosetta mode, as I don't believe all the dependencies are ported yet.

Florisvdf, 2021-04-10:
Solved! 
```
(base) floris@Floriss-MBP ~ % ls -l /dev/tty* | grep usb
crw-rw-rw-  1 root    wheel    9,   6 Apr 10 20:01 /dev/tty.usbmodem0000000000001
```
I got the example running, thanks so much!

lurch, 2021-04-10:
@Florisvdf Please re-open this, and that'll serve as a reminder to us that we need to add better Mac instructions to the getting started guide :wink: