raspberrypi/pico-examples #104: SPI PIO SPI Slave example

asmblur, 2021-04-22:
Would it be possible to get an example of implementing a SPI Slave device in PIO?

aallan, 2021-04-23:
> Would it be possible to get an example of implementing a SPI Slave device in PIO?

See PR #101 for an example.

lurch, 2021-04-23:
@aallan I believe #101 is implementing an SPI Slave using the SPI hardware interface, rather than using PIO as this issue is asking for?

aallan, 2021-04-23:
> @aallan I believe #101 is implementing an SPI Slave using the SPI hardware interface, rather than using PIO as this issue is asking for?

Yup. Reopened.

asmblur, 2021-04-23:
Yeah, I need to create a SPI Slave interface that's slightly modified so the SPI peripheral won't do.  I'm trying to get my head around the PIO but I'm struggling a bit and could really use an example that I can learn from.

lurch, 2021-04-23:
This is a reasonable request, but AFAIK all the RPi Engineers are too busy working on other stuff to be able to look at this any time soon.

asmblur, 2021-04-25:
Thanks for the heads-up! I won't hold my breath then.

Wren6991, 2021-06-02:
I just transferred over #115 from pico-sdk, which is related. There is a sketch of a PIO program in there, which might give you a rough idea:

```
.program clocked_input_chip_select

; Sample bits using an external clock, and push groups of bits into the RX FIFO.
; - IN pin 0 is the data pin
; - IN pin 1 is the clock pin
; - JMP pin is the chip select
; - Autopush is enabled, threshold 8
;
; This program waits for chip select to be asserted (low) before it begins
; clocking in data. Whilst chip select is low, data is clocked continuously. If
; chip select is deasserted part way through a data byte, the partial data is
; discarded. This makes use of the fact a mov to isr clears the input shift
; counter.

flush:
    mov isr, null         ; Clear ISR and input shift counter
    jmp check_chip_select ; Poll chip select again

.wrap_target
do_bit:
    wait 0 pin 1          ; Detect rising edge and sample input data
    wait 1 pin 1          ; (autopush takes care of moving each complete
    in pins, 1            ; data byte to the FIFO)
check_chip_select:
    jmp pin, flush        ; Bail out if we see chip select high
.wrap
```

ThadHouse, 2022-09-22:
https://github.com/ThadHouse/picocolorsensorspi/blob/main/pio_spi/

If anybody still needs this, I built a setup that can do both input and output in peripheral mode over PIO. Basically built for a single mode of SPI, but would be easy enough to manually switch to other modes. Also uses DMA for the transactions, so they can be bigger then a single PIO FIFO.

rspber, 2023-01-28:
https://github.com/rspber/tsdesktop/tree/main/protocols/pio_spi

I have a question about pio_spi working as master. Writing works fine. But reading seems to alternate colors and introduces spontaneous errors. The read speed is also important and works like it works at 2 MHz while the read speed in SPI goes up to 40 MHz.

rspber, 2023-03-29:
PIO reading problem resolved. Thanks.

03jayashree, 2025-04-02:
> PIO reading problem resolved. Thanks.

Can you please give me the github link or some code reference to solve the spi-pio reading problem I am also facing the same issue.