raspberrypi/pico-examples #105: Pico parallel pio

siriokds, 2021-04-24:
Can PIO state machines read/write 8+ pins at time? If not, is there available any tip to handle this?
Thanks.

lurch, 2021-04-24:
Looks like this can be done with the `IN` instruction, with `PINS` as the source? Just set the `Bit count` to the number of pins you need to read?
(and similarly with `OUT` for writing pins)

siriokds, 2021-04-24:
> Looks like this can be done with the `IN` instruction, with `PINS` as the source? Just set the `Bit count` to the number of pins you need to read?
> (and similarly with `OUT` for writing pins)

So I can read/write or change pin direction of 32 pins at once. Did I get it right?

lurch, 2021-04-24:
I believe so, yes (although there's only 26 GPIO pins available on Raspberry Pi Pico).

siriokds, 2021-04-24:
> I believe so, yes (although there's only 26 GPIO pins available on Raspberry Pi Pico).

Thank you.

closed 2021-04-24 by siriokds