raspberrypi/pico-examples #106: PIO for 4MHz Bus interface

siriokds, 2021-04-25:
How many nanoseconds pass between the reception of a data from the PIO and the core that has to manage the data? I'd like to know if a Rpi Pico can handle a slow (4MHz) bus interface.

lurch, 2021-04-26:
The [logic analyser example](https://github.com/raspberrypi/pico-examples/blob/master/pio/logic_analyser/logic_analyser.c) says "1 to 32 pins can be captured, at a sample rate no greater than system clock frequency.".
The Rapsberry Pi Pico runs at a system clock frequency of 133MHz, so using the PIO's divider to slow the SM clock to 4MHz shouldn't be much problem :wink:

siriokds, 2021-04-26:
> The [logic analyser example](https://github.com/raspberrypi/pico-examples/blob/master/pio/logic_analyser/logic_analyser.c) says "1 to 32 pins can be captured, at a sample rate no greater than system [...]

Ok, great! Pls, check this scenario: PIO read 32 bit, send it to TX FiFO, a dedicated core read it, make some stuffs and return a value to TX FIFO in order to let PIO to set gpio. Don't all of these operations have excessive latency?

lurch, 2021-04-26:
Sorry, I'm not going to "do your homework for you". Part of the fun of microcontrollers is experimenting for yourself, and seeing what you can/can't get working! :wink:

lurch, 2021-04-26:
For further questions of this type, please ask for help on the https://www.raspberrypi.org/forums/

closed 2021-04-26 by aallan