raspberrypi/pico-examples #113: Error building hello_usb example

embeddedpenguin, 2021-05-20:
I am not super comfortable with cmake, and I'm pretty sure this is why I'm having trouble. I am using this as my CMakeLists.txt:

```
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

project(hello_usb)

pico_sdk_init()

if (TARGET tinyusb_device)
    add_executable(hello_usb
            hello_usb.c
            )

    # Pull in our pico_stdlib which aggregates commonly used features
    target_link_libraries(hello_usb pico_stdlib)

    # enable usb output, disable uart output
    pico_enable_stdio_usb(hello_usb 1)
    pico_enable_stdio_uart(hello_usb 0)

    # create map/bin/hex/uf2 file etc.
    pico_add_extra_outputs(hello_usb)

    example_auto_set_url(hello_usb)
elseif(PICO_ON_DEVICE)
    message(WARNING "not building hello_usb because TinyUSB submodule is not initialized in the SDK")
endif()
```

I get the following error when running cmake:
```
CMake Error at CMakeLists.txt:24 (example_auto_set_url):
  Unknown CMake command "example_auto_set_url".
```

I'm not sure what function this is trying to call or if this is native to cmake or what. Any guidance would help a lot.

embeddedpenguin, 2021-05-20:
When commenting this out and building, everything compiles but the com port never enumerates from the pico

lurch, 2021-05-20:
> Unknown CMake command "example_auto_set_url".

That's declared in https://github.com/raspberrypi/pico-examples/blob/develop/example_auto_set_url.cmake which gets included into the top-level CMakeLists.txt [here](https://github.com/raspberrypi/pico-examples/blob/develop/CMakeLists.txt#L15)

You don't need it, it just sets one of the strings that gets displayed by "pictool info".

Have you tried reading through Chapter 8 of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf ? :smiley:

lurch, 2021-05-20:
> A lot of the build process is confusing for rp2040.

The easiest approach is to probably work your way through the examples / quickstart-instructions first, before you start creating your own projects :slightly_smiling_face: 
It seems confusing at first, but the build process actually hides a lot of complexity for you.

embeddedpenguin, 2021-05-20:
> > A lot of the build process is confusing for rp2040.
> 
> The easiest approach is to probably work your way through the examples / quickstart-instructions first, before you start creating your own projects 
> It seems confusing at first, but the build process actually hides a lot of complexity for you.

I'm used to just making a makefile and starting with a text editor and using a toolchain and the datasheet to start from scratch. =P

lurch, 2021-05-20:
> I'm used to just making a makefile and starting with a text editor and using a toolchain and the datasheet to start from scratch. =P

Sure, you can do that if you really want to :wink:  https://github.com/raspberrypi/pico-feedback/issues/157