raspberrypi/pico-examples #114: PIO WS2812 example - timing problem

steve-moorby, 2021-05-24:
I was looking at the WS2812 driver to understand how it worked and came across some problems with the timing. The driver is described in section 3.2.2 of the C/C++ SDK. I understood the logic of the program but couldn't see how it met the timing specifications of the WS2812. That was where I discovered the timings in the code and in the description do not match. In the code the timings are T1=2, T2=5 and T3=3. Reading the description and looking at figure 5 and 6 I get timings of T1=4, T2=3 and T3=3. 

There are two times to get right, the short time 400ns and the long time 850ns. Their sum is 1250ns which satisfies the period. Neither the values in the code nor the description seem correct. I calculate T1=8, T2=9 and T3=8 with 50ns clock period. The resolution's finer but still in the range of the delays of the state machine.

			       T3	T1	T2	T short	T long
	code		3	2	5	250ns	875ns
	description	3	4	3	500ns	875ns
	ideal		8	8	9	400ns	850ns

The example code does work so for my sanity I checked with a scope and I do see the calculated timings.  Presumably it's working because the WS2812 is forgiving.

I also simplified the state machine code avoiding jumps by putting the shifted out bit onto the output line.

.program ws2812

.define public T0 8
.define public T1 9

.wrap_target
    set pins, 0    [T0 - 2]	   ; start off low
    out x, 1                       ; get a single bit from the shift register
    set pins, 1    [T0 - 1]        ; start the high
    mov pins, x    [T1 - 1]        ; continue waveform high or low according to bit
.wrap

Note there are corresponding changes to the C and Python code for state machine configuration.

lurch, 2021-05-25:
This might count as a "duplicate" of https://github.com/raspberrypi/pico-feedback/issues/121 ? :shrug:

aallan, 2021-05-25:
Discussion probably better belongs over the other repo anyway, so raspberrypi/pico-feedback#121 might be the place to talk about it.

closed 2021-05-25 by aallan

steve-moorby, 2021-05-25:
Agreed it's a duplicate.  After reading that thread and seeing the variation in timings I don't think the 50ns resolution is important. I've gone for the 1/3, 2/3 duty cycle giving timings of 42ns and 83ns and it works fine although I can't say what actual driver chips I've got.

.program ws2812

.define public T0 2
.define public T1 2

.wrap_target
set pins, 0 [T0 - 2] ; start off low
out x, 1 ; get a single bit from the shift register
set pins, 1 [T0 - 1] ; start the high
mov pins, x [T1 - 1] ; continue waveform high or low according to bit
.wrap

Whatever the outcome of this one it's a good illustration that the RP2040 PIO can meet peripheral timings requirements with no burden to the CPU.

aallan, 2021-05-25:
Thanks for following up and confirming!

steve-moorby, 2021-05-25:
One last tinker!  Using the relaxed timings of 1/3, 2/3 duty cycle allows a switch back to side set resulting in a very minimalist solution.

    out x, 1    side 0   ; get a single bit from the shift register and start off low
    nop         side 1   ; start the high
    mov pins, x          ; continue waveform high or low according to bit