raspberrypi/pico-examples #123: Is "IRQ 4 REL" a Valid PIO Instruction?

soundpaint, 2021-06-13:
Hi,

pio/uart_rx/uart_rx.pio contains the following PIO instruction (line 57):

irq 4 rel

This instruction triggers an assertion on my PIO emulator.  And I think the emulator is right, because RP2040 datasheet, Sect. 3.4.9.2. says:

"If the MSB is set, the state machine ID (0...3) is added to the IRQ index, by way of modulo-4 addition on the two LSBs."

That is, the result will always be in the range 0…3, such that "rel" make sense only in combination with IRQ numbers 0…3.  Or is "irq 4 rel" considered to be equivalent with "irq 0 rel", just ignoring the third-lowest bit?

Or should the spec read instead:

"If the MSB is set, the state machine ID (0...3) is added to **the lowest two LSBs of** the IRQ index, by way of modulo-4 addition on the two LSBs."?

Then, you would have two circular groups of IRQs, namely IRQs 0...3, and IRQs 4…7.

Just being somewhat confused,
Jürgen

nonchip, 2021-06-27:
> Or is "irq 4 rel" considered to be equivalent with "irq 0 rel", just ignoring the third-lowest bit?

isn't that what "modulo-4" means?

soundpaint, 2021-06-27:
Ah, ok, thank you very much!  Then I will go for the "circular groups", as I called them. :-)

I just considered the wording of the spec ambiguous for this specific case, but maybe my English skills are simply not optimal.
From the wording "by way of modulo-4 addition" I concluded, that the result of this operation (i.e. the resulting IRQ index, I thought) will be in the range <= 3.

closed 2021-06-27 by soundpaint