raspberrypi/pico-examples #124: Conflicting Code Comment in spi.pio

soundpaint, 2021-06-15:
pico-examples/pio/spi/spi.pio, lines 17-20:

; Pin assignments:
; - SCK is side-set pin 0
; - MOSI is OUT pin 0
; - MISO is IN pin 0

However, spi_loopback.c:

#define PIN_SCK 18
#define PIN_MOSI 16
#define PIN_MISO 16 // same as MOSI, so we get loopback

Actually, I think choosing SCK to use the same pin as MOSI or MISO does not make much sense.  That is, spi.pio looks wrong to me, while spi_loopback.c  looks fully ok.  Is my observation correct?
Or maybe the values in spi.pio are meant to be relative to the pin base?
Maybe the comments in spi.pio should be removed or clarified? :-)

lurch, 2021-06-15:
See section "3.5.6. GPIO Mapping" of the RP2040 datasheet :wink:

soundpaint, 2021-06-16:
> See section "3.5.6. GPIO Mapping" of the RP2040 datasheet

Mmmh, I do not understand.
spi_loopback.c configures the mapping, just like described in 3.5.6., to pins 16 and 18.  So far, so well.
But what is the meaning of "SCK is side-set pin 0" or "MOSI is OUT pin 0"?  Since there is no pin base value that you could pass to either PIO instruction IN or OUT, it is implicitly always 0, when assuming that these values refer to base pin before GPIO mapping?  But then there is no point in those comments?
Now, I am even more confused...

lurch, 2021-06-16:
There's separate BASE values for each of SET, OUT, SIDE-SET and IN. These are configured on a per-state-machine basis (see https://github.com/raspberrypi/pico-examples/blob/master/pio/spi/spi.pio#L49-L51 and Figure 48 in the RP2040 datasheet) as this allows multiple state-machines to share the same instruction-memory, just with different per-SM config.
So for this SPI PIO example, when the PIO code (in `spi.pio`) does OUT to "out pin 0" (which the comment identifies as being used for MOSI), it actually modifies `PINCTRL_OUT_BASE + 0`, and since `spi_loopback.c` initialises PINCTRL_OUT_BASE (for this SM) to 16, this has the affect of modifying GPIO16. Similarly, when the PIO code does IN from "in pin 0" (which the comment identifies as being used for MISO), it actually reads from `PINCTRL_IN_BASE + 0`, and since `spi_loopback.c` initialises PINCTRL_IN_BASE (for this SM) to 16, this has the affect of reading from GPIO16. PIO allows pins set as 'output' to also have their value read by an IN command, and this is what achieves the loopback functionality.

spi_loopback.c could (theoretically) instead initialise two copies of the spi.pio SM, with the MOSI of one SM (i.e. PINCTRL_OUT_BASE) set to GPIO19 and the MISO (i.e. PINCTRL_IN_BASE) set to GPIO16, and the MOSI of the other SM set to GPIO12 and MISO set to GPIO15; but the example would then require you to connect jumper wires between pins GPIO19 + GPIO15 and GPIO16 + GPIO12, rather than using the "internal loopback" that the example currently uses.
(And this is an oversimplification, since the CLK signal would also need to be connected between the two SMs, and the code in spi.pio actually only handles SPI master mode and not SPI slave mode)

I'm not sure how well I explained that (I'm not a PIO expert), but if it still doesn't make sense I can only suggest that you read through the PIO chapter in the RP2040 datasheet a couple more times.

soundpaint, 2021-06-16:
I can completely follow your excellent explanation, and it fully matches my understanding.  Still, how could you achieve, say, e.g. a PINCTRL_OUT_BASE + 5, when count=1?  The PIO OUT instruction has a bit count parameter, but no bit base parameter.  Hence, for OUT instructions, bit base (prior to GPIO mapping) is _always_ 0, as far as I understand.  So, I think there is no point in explicitly stating the "0".
But it also does not harm.  Before wasting more time, feel free to close this issue; its priority is from my point of view neglectibly low. :-)

soundpaint, 2021-06-16:
By the way, you write:

    does OUT to "out pin 0"

This is not a valid PIO instruction.  Sect. 3.4.5.3. says:

    out <destination>, <bit_count>
    where:
    <destination> Is one of the destinations specified above.
    <bit_count> Is a value (see Section 3.3.2) specifying the number of bits to shift (valid range 1-32)

:-)

lurch, 2021-06-16:
> This is not a valid PIO instruction.

I did warn you I wasn't a PIO expert :wink:  I may have got some of my terminology / usage wrong, but hopefully you now understand what's going on?

> Still, how could you achieve, say, e.g. a PINCTRL_OUT_BASE + 5, when count=1?

Having a quick skim through the RP2040 datasheet, if looks like if you've configured OUT COUNT to 1, but then do `OUT PINS, 5` then it probably reads 5 bits from the OSR, sets PINCTRL_OUT_BASE to the value of the first bit, and then discards the other 4 bits. Obviously you'd increase the count if you want to write to more pins at once. And if you only wanted to write to a single pin rather than all the OUT pins at once, I guess that's where the flexibility of SET or SIDE-SET come into play.

> The PIO OUT instruction has a bit count parameter, but no bit base parameter.

Correct, the SM's OUT_BASE is set via the PINCTRL_OUT_BASE register (or more conveniently, using the [pio_sm_set_out_pins](https://raspberrypi.github.io/pico-sdk-doxygen/group__sm__config.html#ga7f0190e662c5b6e24d91044b42010ffc) function.

soundpaint, 2021-06-17:
> sets PINCTRL_OUT_BASE to the value of the first bit, and then discards the other 4 bits

Yes, this is also my understanding.

> Obviously you'd increase the count if you want to write to more pins at once.

Yes, if you want to write to more pins at once, but no just because the "OUT PINS" instruction contains a higher bit count value: The data in the OSR may be encoded in a way such that it contains e.g. additional bits for bookkeeping or flow control, i.e. it contains bits that are not intended to actually be asserted to output pins.  In this case, you would only output only some of the bits and discard all others with the shift of the OSR.

Anyway, meanwhile I had a look at the spi_cpha0_cs, and then, at least for, it _really_ gets clear how to read the comment.
In the spi_cpha0_cs, it reads:

  ; Pin assignments:
  ; - SCK is side-set bit 0
  ; - CSn is side-set bit 1
  ; - MOSI is OUT bit 0 (host-to-device)
  ; - MISO is IN bit 0 (device-to-host)

and

  .side_set 2

That is, here, "SCK is side-set bit 0" means "SCK is side-set bit 0 of the _two_ side-set bits" and "CSn is side-set bit 1" means "CSn is side-set bit 1 of the _two_ side-set bits".  Ok, this now clear.
And then, the comments in the spi_cpha0 is just formulated in a way such that it is consistent with the spi_cpha0_cs example.
Ok, so the comments in the spi_cpha0  _are_ correct, but (at least for me) _only_ after reading the comments in the spi_cpha0_cs example...

But as I already said, it's just a comment.  We already spent way to much time for this (almost non-)issue.  I now think the comment is correct, but misleading, as long as you do not have seen the spi_cpha0_cs example.  But feel to close this issue, I won't complain. :-)

lurch, 2021-06-17:
@kilograham Can you close this please?

closed 2021-06-18 by soundpaint