raspberrypi/pico-examples #125: spi.pio: Should Also Invert Input When CPOL=1?

soundpaint, 2021-06-16:
In spi.pio, assume CPOL=1.

spi_loopback.c says:

// ... Any data written into the state machine's TX FIFO should then be
// serialised, deserialised, and reappear in the state machine's RX FIFO.

However, spi.pio says:

    // The pin muxes can be configured to invert the output (among other things
    // and this is a cheesy way to get CPOL=1
    gpio_set_outover(pin_sck, cpol ? GPIO_OVERRIDE_INVERT : GPIO_OVERRIDE_NORMAL);

while there is no call to function  gpio_set_inover().
That is, the signal will be inverted when being output (via out override), but not inverted back when looped back.  So, the expected input in the RX FIFO will be the inverted bit pattern of the TX FIFO values?
Or should a call to gpio_set_inover() be added to spi.pio for correct handling of polarity?

Just being a lttle bit confused... :-)

lurch, 2021-06-16:
CPOL = Clock POLarity, which needs to match on both the "master" and "slave" sides of the SPI connection. And for this loopback connection, since both "sides" are running on the same state-machine, they'll always be using the same clock-polarity setting.

soundpaint, 2021-06-16:
Aah, ok,  It turned out, that in my (minimally adapted) working example, the gpio_set_outover() was erroneously applied onto the MOSI pin rather than onto the SCK pin, which explains why I saw the bits being inverted in the RX FIFO when looped back.
=> Completely my fault!  I'm sorry for wasting your time for this issue!

closed 2021-06-16 by soundpaint