raspberrypi/pico-examples #128: Building hello_usb fails

exit-failure, 2021-06-25:
I'm trying to build all examples in Windows. Unfortunately the process fails when it tries to link hello_usb. It reports that it can't find hello_usb.c.obj but it managed to build it and I can see that the file exists in the specified directory.

```
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/hello_usb.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_uart/uart.c.obj
[  2%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_divider/divider.S.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_time/time.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_time/timeout_helper.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_timer/timer.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_sync/sem.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_sync/lock_core.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_sync/mutex.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_sync/critical_section.c.obj
[  2%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_util/datetime.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_util/pheap.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/common/pico_util/queue.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_clocks/clocks.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_pll/pll.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_vreg/vreg.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_watchdog/watchdog.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/hardware_xosc/xosc.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_divider/divider.S.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_double/double_aeabi.S.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_double/double_init_rom.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
[  3%] Building ASM object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
[  3%] Building CXX object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_stdio/stdio.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_stdio_usb/reset_interface.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/dcd_rp2040.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/device/usbd.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/device/usbd_control.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/audio/audio_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/cdc/cdc_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/dfu/dfu_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/dfu/dfu_rt_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/hid/hid_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/midi/midi_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/msc/msc_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/net/net_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/usbtmc/usbtmc_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/class/vendor/vendor_device.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/tusb.c.obj
[  3%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/lib/tinyusb/src/common/tusb_fifo.c.obj[  4%] Building C object hello_world/usb/CMakeFiles/hello_usb.dir/D_/git/pico-sdk/src/rp2_common/pico_fix/rp2040_usb_device_enumeration/rp2040_usb_device_enumeration.c.obj
[... cut ...]
AR10B2~1.EXE: error: ´╗┐CMakeFiles/hello_usb.dir/hello_usb.c.obj: No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```

lurch, 2021-06-28:
Which compiler are you using? The ARM compiler/linker included with Visual Studio seems to have problems, but the compiler downloaded from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads (as detailed in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf ) works okay.

yunne, 2021-06-29:
I also have that problem. I use the compiler from that link and it's still error, but in my macOS it's work fine. I followed the instructions in the datasheet every step. It confused me too.

aallan, 2021-06-29:
This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/viewforum.php?f=143), so I’m closing this issue. But please follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation/rp2040/getting-started/).

closed 2021-06-29 by aallan

SpamWall, 2021-08-17:
The pico forum (https://www.raspberrypi.org/forums/viewforum.php?f=143) is lame.  No replies to problems such as these.  No help.  Please don't keep directing us there unless it is helpful.

mikessut, 2024-04-16:
I sure would love to know how to resolve this as I spent the better part of a day trying to figure it out.  I think it's a matter of nmake at the gcc compiler (even though cmake said it was using the gcc compiler).

lurch, 2024-04-16:
Since this issue was first created, we've released https://github.com/raspberrypi/pico-setup-windows (which has its own [tutorial](https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md) ), and also updated https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf