raspberrypi/pico-examples #141: Error building hello_usb on Windows 10

AtaKhiabani, 2021-07-12:
I have taken the steps in chapter 9.2 of "Getting started with raspberry pi Pico" and set up the Pico for my Windows machine and I can build the "blink" example successfully. But I am getting the folloeing error when I build the hello_usb example:

`AR10B2~1.EXE: error: ∩╗┐CMakeFiles/hello_usb.dir/hello_usb.c.obj: No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.`

I have ran the following command inside pico-sdk directory with no luck,
`git submodule update --init`

I have also added this path:
`C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30037\bin\HostX86\x86`
to my "path" variable both for user variables and system variables, but still I get error building hello_usb example. Any help is greatly appreciated!

lurch, 2021-07-12:
Duplicate of #128 ?

khowling, 2021-07-25:
+1.  Running ```nmake``` on the pico examples on Windows 10 as per PDF instructions.  the ```blink.c.obj``` file actually exists in ```pico-examples\build\blink\CMakeFiles\blink.dir```.  So I'm sure its a directory thing on Windows 10 - See the 2 odd characters at the start of the path in the error!

[image]

Any help will be greatly received :)

lurch, 2021-07-25:
Hmmm, searching for those strange characters finds https://superuser.com/questions/601282/%CC%81-is-not-recognized-as-an-internal-or-external-command/804342 which sounds like a plausible explanation. However I'm not sure which part of the build-process is (incorrectly) creating files with the UTF8 BOM, nor why this is only affecting a very small minority of users :confused: 
@kilograham is more familiar with the build-process so he might have some ideas?

@AtaKhiabani and @khowling What version of CMake do you have installed? Do you have non-ASCII characters somewhere in your directory path?

khowling, 2021-07-25:
Followed the Windows10 install instructions today, so:
cmake version 3.21.0, downloaded today ```cmake-3.21.0-windows-x86_64.msi```
directory path (nothing appears odd): ```C:\Users\khowling\pico\pico-examples```
```git version 2.21.0.windows.1```
```Visual Studio Build Tools 2019 - 16.10.4```
```git clone -b master https://github.com/raspberrypi/pico-examples.git``` ran today (wiped and re-cloned twice to try to resolve)
.

AtaKhiabani, 2021-07-25:
@khowling and @lurch,
I tried different versions of CMake and also different versions of Windows 10 SDK but couldn't fix this issue. Finally, after one week of trying to run the examples, I found out about this really helpful tutorial that uses MinGW instead of nmake:
https://shawnhymel.com/2096/how-to-set-up-raspberry-pi-pico-c-c-toolchain-on-windows-with-vs-code/
Using this I am able to build and run all the pico examples. I suggest using this and not wasting your time with nmake.

khowling, 2021-07-26:
I was using WSL2, the toolchain works great on WSL2 .  Only issue is access to the USB device, the picotool etc doesn't work on WSL2, so moved to Windows to try to get a more automated workflow

kilograham, 2021-09-12:
https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=316310 apparently cmake 3.20.X _does_ fix this issue

xenois, 2021-11-27:
Windows has some weird long path restrictions which cannot be disabled even with regedit. I just created a [project](https://github.com/xenois/pico-dev) for using Docker to build everything (debugging included). Saves you from configuring everything new in different OS.

CrackedP0t, 2022-01-04:
I had this problem with CMake 3.22.1, but downgrading to CMake 3.20.5 as was suggested in [this thread](https://forums.raspberrypi.com/viewtopic.php?f=145&t=316310) fixed it.

paramthiru, 2022-01-19:
Thank you @CrackedP0t! After facing the same issue with CMake 3.22.1, downgrading to CMake 3.20.5 worked.

raimuucka, 2022-01-22:
> https://www.raspberrypi.org/forums/viewtopic.php?f=145&t=316310 apparently cmake 3.20.X _does_ fix this issue

Worked for me too. Uninstalled my current CMake (3.22.1). Installed 3.20.5 and deleted build content, then repeated 
`cmake -G "NMake Makefiles" ..`
and then 
`nmake`

insane-kane, 2022-01-29:
This appears to be a bug with CMake 3.21+ for Windows, not with the Raspberry Pi Pico SDK or files.  I was able to replicate this issue with an entirely new project, still using the pico SDK albeit.

`[build] [100%] Linking CXX executable blink-1.elf
[build] AR10B2~1.EXE: error: ∩╗┐CMakeFiles/blink-1.dir/blink-1.c.obj: No such file or directory
[build] NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102021~1.10\bin\AR10B2~1.EXE' : return code '0x1'
[build] Stop.`

insane-kane, 2022-01-29:
> This appears to be a bug with CMake 3.21+ for Windows, not with the Raspberry Pi Pico SDK or files. I was able to replicate this issue with an entirely new project, still using the pico SDK albeit. [...]

Downgrading to CMake version 3.20.6 corrected the issue.

nhorlock, 2022-08-06:
Using CMake 3.24.0 (from cmake.org) also appears to fix this.

Talha-Babar, 2024-05-14:
i am using CMake 3.29.3 and still facing the issue in building the pio examples

The error is:

CMake Error at CMakeLists.txt:18 (pico_add_extra_outputs):
  Unknown CMake command "pico_add_extra_outputs".

-- Configuring incomplete, errors occurred!

lurch, 2024-05-14:
@Talha-Babar You might want to try using https://github.com/raspberrypi/pico-setup-windows ?