raspberrypi/pico-examples #152: error building blink on windows

Lite5h4dow, 2021-08-14:
```
AR10B2~1.EXE: error: ´╗┐CMakeFiles/blink.dir/blink.c.obj: No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```
i get the following error attempting to build the examples. not sure why.

lurch, 2021-08-14:
I see you have a strange `╗┐` in your output above, so this is probably the same problem as #141 
Which unfortunately we don't know how to solve yet :confused:

Lite5h4dow, 2021-08-15:
i dont think its those characters because I ran the same thing in powershell and got this
[image]

kilograham, 2021-08-15:
yeah, i don't think that tells us anything as powershell is just displaying the 3 byte UTF8 sequence  (ASCII´╗┐) as a zero width character.

The question is how did the characters get into the start of the filename? And why does it only happen for some people.

Can you try building with ninja instead

```
mkdir ninja_build
cd ninja_build
cmake -G Ninja ..
ninja
``` 
to see if NMake is the issue)?

Lite5h4dow, 2021-08-16:
@kilograham i dont have ninja installed, could you point me in the right direction?

Lite5h4dow, 2021-08-16:
@kilograham 
i get he following
```
[15/4743] Performing build step for 'ELF2UF2Build'
[1/2] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.obj
C:/Users/storm/Dev/pico-sdk/tools/elf2uf2/main.cpp:328:16: warning: 'fopen' is deprecated: This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [-Wdeprecated-declarations]
    FILE *in = fopen(in_filename, "rb");
               ^
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\ucrt\stdio.h:212:20: note: 'fopen' has been explicitly marked deprecated here
    _Check_return_ _CRT_INSECURE_DEPRECATE(fopen_s)
                   ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:320:55: note: expanded from macro '_CRT_INSECURE_DEPRECATE'
        #define _CRT_INSECURE_DEPRECATE(_Replacement) _CRT_DEPRECATE_TEXT(    \
                                                      ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:310:47: note: expanded from macro '_CRT_DEPRECATE_TEXT'
#define _CRT_DEPRECATE_TEXT(_Text) __declspec(deprecated(_Text))
                                              ^
C:/Users/storm/Dev/pico-sdk/tools/elf2uf2/main.cpp:334:17: warning: 'fopen' is deprecated: This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [-Wdeprecated-declarations]
    FILE *out = fopen(out_filename, "wb");
                ^
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\ucrt\stdio.h:212:20: note: 'fopen' has been explicitly marked deprecated here
    _Check_return_ _CRT_INSECURE_DEPRECATE(fopen_s)
                   ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:320:55: note: expanded from macro '_CRT_INSECURE_DEPRECATE'
        #define _CRT_INSECURE_DEPRECATE(_Replacement) _CRT_DEPRECATE_TEXT(    \
                                                      ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:310:47: note: expanded from macro '_CRT_DEPRECATE_TEXT'
#define _CRT_DEPRECATE_TEXT(_Text) __declspec(deprecated(_Text))
                                              ^
2 warnings generated.
[2/2] Linking CXX executable elf2uf2.exe
[16/4743] Performing build step for 'PioasmBuild'
FAILED: adc/dma_capture/pioasm/src/PioasmBuild-stamp/PioasmBuild-build C:/Users/storm/Dev/pico-examples/ninja_build/adc/dma_capture/pioasm/src/PioasmBuild-stamp/PioasmBuild-build
cmd.exe /C "cd /D C:\Users\storm\Dev\pico-examples\ninja_build\pioasm && C:\Users\storm\scoop\apps\cmake\3.21.1\bin\cmake.exe --build ."
[1/10] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.obj
FAILED: CMakeFiles/pioasm.dir/gen/lexer.cpp.obj
C:\Users\storm\scoop\shims\clang++.exe  -IC:/Users/storm/Dev/pico-sdk/tools/pioasm -IC:/Users/storm/Dev/pico-sdk/tools/pioasm/gen -g -Xclang -gcodeview -O0 -D_DEBUG -D_DLL -D_MT -Xclang --dependent-lib=msvcrtd -std=gnu++14 -MD -MT CMakeFiles/pioasm.dir/gen/lexer.cpp.obj -MF CMakeFiles\pioasm.dir\gen\lexer.cpp.obj.d -o CMakeFiles/pioasm.dir/gen/lexer.cpp.obj -c C:/Users/storm/Dev/pico-sdk/tools/pioasm/gen/lexer.cpp
lexer.ll:38:10: fatal error: 'unistd.h' file not found
#include <unistd.h>
         ^~~~~~~~~~
1 error generated.
[2/10] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.obj
[3/10] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.obj
[4/10] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.obj
[5/10] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.obj
[6/10] Building CXX object CMakeFiles/pioasm.dir/main.cpp.obj
[7/10] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.obj
[8/10] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.obj
[9/10] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.obj
C:/Users/storm/Dev/pico-sdk/tools/pioasm/pio_assembler.cpp:385:47: warning: 'fopen' is deprecated: This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details. [-Wdeprecated-declarations]
    FILE *out = destination == "-" ? stdout : fopen(destination.c_str(), "w");
                                              ^
C:\Program Files (x86)\Windows Kits\10\Include\10.0.19041.0\ucrt\stdio.h:212:20: note: 'fopen' has been explicitly marked deprecated here
    _Check_return_ _CRT_INSECURE_DEPRECATE(fopen_s)
                   ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:320:55: note: expanded from macro '_CRT_INSECURE_DEPRECATE'
        #define _CRT_INSECURE_DEPRECATE(_Replacement) _CRT_DEPRECATE_TEXT(    \
                                                      ^
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include\vcruntime.h:310:47: note: expanded from macro '_CRT_DEPRECATE_TEXT'
#define _CRT_DEPRECATE_TEXT(_Text) __declspec(deprecated(_Text))
                                              ^
1 warning generated.
ninja: build stopped: subcommand failed.
[17/4743] No install step for 'ELF2UF2Build'
ninja: build stopped: subcommand failed.
```

kilograham, 2021-08-16:
You seem to be building there with clang

```
C:\Users\storm\scoop\shims\clang++.exe
```

definitely make sure you don't have `CC` or `CXX` pointing at something weird in your environment. you should be running from a Visual Studio command shell (I forget what it is called)

i would hope ninja build vs nmake build would pick up the same system compiler. can you post the raw output of a clean (fresh build directory) build with nmake too.

Lite5h4dow, 2021-08-17:
here is the full output as requested
```
C:\Users\storm\Dev\pico-examples\build> cmake -G "NMake Makefiles" ..
Using PICO_SDK_PATH from environment ('c:/users/storm/dev/pico-sdk')
PICO_SDK_PATH is C:/Users/storm/Dev/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2021.07/bin/arm-none-eabi-gcc.exe
Defaulting PICO target board to pico since not specified.
Using board configuration from C:/Users/storm/Dev/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: C:/Users/storm/scoop/shims/python3.exe (found version "3.9.6") found components: Interpreter
TinyUSB available at C:/Users/storm/Dev/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; adding USB support.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/storm/Dev/pico-examples/build

C:\Users\storm\Dev\pico-examples\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.29.30133.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Scanning dependencies of target bs2_default
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] Linking ASM executable bs2_default.elf
[  0%] Built target bs2_default
[  0%] Generating bs2_default.bin
[  0%] Generating bs2_default_padded_checksummed.S
[  0%] Built target bs2_default_padded_checksummed_asm
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] No download step for 'ELF2UF2Build'
[  0%] No update step for 'ELF2UF2Build'
[  0%] No patch step for 'ELF2UF2Build'
[  0%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is MSVC 19.29.30133.0
-- The CXX compiler identification is MSVC 19.29.30133.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx86/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/storm/Dev/pico-examples/build/elf2uf2
[  0%] Performing build step for 'ELF2UF2Build'

Microsoft (R) Program Maintenance Utility Version 14.29.30133.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.obj
main.cpp
C:\Users\storm\Dev\pico-sdk\tools\elf2uf2\main.cpp(328): warning C4996: 'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Users\storm\Dev\pico-sdk\tools\elf2uf2\main.cpp(334): warning C4996: 'fopen': This function or variable may be unsafe. Consider using fopen_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
[100%] Linking CXX executable elf2uf2.exe
[100%] Built target elf2uf2
[  0%] No install step for 'ELF2UF2Build'
[  0%] Completed 'ELF2UF2Build'
[  0%] Built target ELF2UF2Build
Scanning dependencies of target blink
[  0%] Building C object blink/CMakeFiles/blink.dir/blink.c.obj
[  0%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[... 30 lines cut ...]
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_float/float_aeabi.S.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_float/float_init_rom.c.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_float/float_math.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_float/float_v1_rom_shim.S.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_malloc/pico_malloc.c.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_mem_ops/mem_ops_aeabi.S.obj
[  1%] Building ASM object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_standard_link/crt0.S.obj
[  1%] Building CXX object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp.obj
[  1%] Building C object blink/CMakeFiles/blink.dir/C_/Users/storm/Dev/pico-sdk/src/rp2_common/pico_standard_link/binary_info.c.obj
[... cut ...]
AR10B2~1.EXE: error: ´╗┐CMakeFiles/blink.dir/blink.c.obj: No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102021~1.07\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.

C:\Users\storm\Dev\pico-examples\build>
```

Lite5h4dow, 2021-08-17:
building with ninja in the "Developer console for VS 2019" does appear to work however

Lite5h4dow, 2021-08-17:
ive settled on using the Ninja pipeline in VSCode via CMake.

closed 2021-08-17 by Lite5h4dow

krisradio, 2021-12-18:
Just want to comment that I am seeing the same issue, using nmake: 

```
AR10B2~1.EXE: error: ∩╗┐CMakeFiles/blink.dir/blink.c.obj: No such file or directory
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102021~1.10\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.30.30705\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files\Microsoft Visual Studio\2022\Professional\VC\Tools\MSVC\14.30.30705\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
```