raspberrypi/pico-examples #153: Error building hello_usb on Windows 10

HrOlsen, 2021-08-18:
I've followed this guide step by step:
https://www.element14.com/community/community/raspberry-pi/blog/2021/01/24/working-with-the-raspberry-pi-pico-with-windows

And everything works in "Developer Command Prompt for VS 2019" until the linking where I also have the "╗┐" in front of the directory path with the object files in.
[image]

HrOlsen, 2021-08-18:
I can see that it has been reported previously but never solved.
https://github.com/raspberrypi/pico-examples/issues/152
https://github.com/raspberrypi/pico-examples/issues/141

closed 2021-08-18 by HrOlsen

aallan, 2021-08-18:
I've not read through that guide.

Did you look at Chapter 9 (section 9.2) of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf, which is the **official** guide to how to do this, and worked as of 2 days ago when I last tested it. While Raspberry Pi does not directly support it there is also third-party installer script for Windows 10 that is roughly equivalent of the `pico-setup.sh` script . More details at https://github.com/ndabas/pico-setup-windows.

lurch, 2021-08-18:
@aallan It seems that _some_ users (even when following our PDF instructions) are getting a weird UTF-8 byte-sequence in one of their build-files, and this is causing some part of the build-system to choke. But as of yet neither myself nor @kilograham have been able to figure out why, and we can't reproduce the problem ourselves.

aallan, 2021-08-18:
> …some users (even when following our PDF instructions) are getting a weird UTF-8 byte-sequence in one of their build-files, and this is causing some part of the build-system to choke

Yup. I've had a go at reproducing it. I also can't figure out what's going on, _"Works on my machine™"_ . 🤷‍♂️

aallan, 2021-08-18:
My guess is (was?) that there is a tutorial somewhere on the web that's the root cause of this, it's telling them to do something that mostly but not always works.

lurch, 2021-08-18:
My guess was some kind of character-encoding / localisation / filesystem issue, but who knows :man_shrugging:

kilograham, 2021-08-18:
i have not (nor will be at any time soon) looked in any detail. the UTF-8 byte sequence is a "byte order marking" that Windows/some windows editors like to insert at the top of UTF-8 files when writing them. these seem to get into the file containing filenames to link, and hence presumably the first filename.

Either this only happens sometimes or some versions of ARM gcc can cope with it.

Also, allegedly the problem goes away if you use ninja (or mingw), so nmake is part of the mix

HrOlsen, 2021-08-19:
I'm up and running now ;)
I uninstalled everything and followed this guide:
https://shawnhymel.com/2096/how-to-set-up-raspberry-pi-pico-c-c-toolchain-on-windows-with-vs-code/#Directory_Setup
There is a step for "enable long paths" - i don't know if this is the tweak that fixes the issue.

lurch, 2021-08-19:
@HrOlsen I think it's the fact that it uses mingw that "fixes" it for you. (which Graham also mentions in his comment above)