raspberrypi/pico-examples #186: Problem with bmp280_i2c example

Glitchbone, 2021-11-16:
I'm not sure if it's related to the example code, but the serial output of the bmp280_i2c always shows incorrect values like :
```
Pressure = 36479.457 kPa
Temp. = -15.94 C
```
The good thing is that the values are not constant, if I put the sensor close to a heating source, the values quickly go up.

After some debugging, it seems that the calibration data I get from the BMP280 are wrong, here's a dump of the raw bytes I get from inside bmp280_get_calib_params() after i2c_read_blocking : 

`00 00 00 00 00 2F 94 00 4C 9D C0 7B F3 00 72 74 90 43 A3 2B BA 00 00 6B`

According to the datasheet, it means dig_T1 and dig_T2 are zero but they state that compensation values are never zero...

I tried with 2 different BMP280 (3.3V) modules wired as in the example schematic but the result is the same.

I'm now out of idea :)

Does anyone have an idea on what could cause this issue?

Thanks in advance

Glitchbone, 2021-11-17:
**UPDATE**

I noticed some strange behavior while debugging, if I read the whole memory of the BMP280, I get the correct values, but trying to read starting from 0x88 or even just the chip ID register at 0xD0, the returned bytes are wrong.

The following code inside the main loop : 
```
int w1, r1, w2, r2;

// Read all registers

uint8_t buf1[255] = { 0 };
w1 = i2c_write_blocking(i2c_default, ADDR, (uint8_t*)0x00, 1, true);
r1 = i2c_read_blocking(i2c_default, ADDR, buf1, 255, false);

printf("Full dump (%d bytes written, %d bytes read) :\n", w1, r1);
for (uint8_t i = 0; i < 255; i++) {
    printf("%02X:%02X ", i, buf1[i]);
}

// Read calibration registers only

uint8_t buf2[24] = { 0 };
w2 = i2c_write_blocking(i2c_default, ADDR, (uint8_t*)0x88, 1, true);
r2 = i2c_read_blocking(i2c_default, ADDR, buf2, 24, false);

printf("\n\nCalibration registers (%d bytes written, %d bytes read) :\n", w2, r2);
for (uint8_t i = 0; i < 24; i++) {
    printf("%02X:%02X ", 0x88+i, buf2[i]);
}
```

will give the following output : 

```
Full dump (1 bytes written, 255 bytes read) :
00:00 01:00 02:00 03:00 04:00 05:00 06:00 07:00 08:00 09:00 0A:00 0B:00 0C:00 0D:00 0E:00 0F:00
10:00 11:00 12:00 13:00 14:00 15:00 16:00 17:00 18:00 19:00 1A:00 1B:00 1C:00 1D:00 1E:00 1F:00
20:00 21:00 22:00 23:00 24:00 25:00 26:00 27:00 28:00 29:00 2A:00 2B:00 2C:00 2D:00 2E:00 2F:00
30:00 31:00 32:00 33:00 34:00 35:00 36:00 37:00 38:00 39:00 3A:00 3B:00 3C:00 3D:00 3E:00 3F:00
40:00 41:00 42:00 43:00 44:00 45:00 46:00 47:00 48:00 49:00 4A:00 4B:00 4C:00 4D:00 4E:00 4F:00
50:00 51:00 52:00 53:00 54:00 55:00 56:00 57:00 58:00 59:00 5A:00 5B:00 5C:00 5D:00 5E:00 5F:00
60:00 61:00 62:00 63:00 64:00 65:00 66:00 67:00 68:00 69:00 6A:00 6B:00 6C:00 6D:00 6E:00 6F:00
70:00 71:00 72:00 73:00 74:00 75:00 76:00 77:00 78:00 79:00 7A:00 7B:00 7C:00 7D:00 7E:00 7F:00
80:85 81:50 82:90 83:69 84:1D 85:3E 86:D0 87:00 88:7B 89:6D 8A:75 8B:67 8C:18 8D:FC 8E:34 8F:8E
90:3F 91:D6 92:D0 93:0B 94:8A 95:04 96:1E 97:01 98:F9 99:FF 9A:8C 9B:3C 9C:F8 9D:C6 9E:70 9F:17
A0:00 A1:00 A2:82 A3:00 A4:00 A5:00 A6:00 A7:00 A8:00 A9:00 AA:00 AB:00 AC:33 AD:00 AE:00 AF:C0
B0:00 B1:54 B2:00 B3:00 B4:00 B5:00 B6:60 B7:02 B8:00 B9:01 BA:FF BB:82 BC:13 BD:71 BE:03 BF:00
C0:0A C1:00 C2:2F C3:FF C4:00 C5:00 C6:00 C7:00 C8:00 C9:00 CA:00 CB:00 CC:00 CD:00 CE:00 CF:00
D0:58 D1:00 D2:00 D3:00 D4:00 D5:00 D6:00 D7:00 D8:00 D9:00 DA:00 DB:00 DC:00 DD:00 DE:00 DF:00
E0:00 E1:00 E2:00 E3:00 E4:00 E5:00 E6:00 E7:00 E8:00 E9:00 EA:00 EB:00 EC:00 ED:00 EE:00 EF:00
F0:00 F1:00 F2:00 F3:00 F4:2F F5:94 F6:00 F7:6B F8:CE F9:80 FA:7E FB:73 FC:00 FD:85 FE:50 

Calibration registers (1 bytes written, 24 bytes read) :
88:00 89:00 8A:00 8B:00 8C:00 8D:2F 8E:94 8F:00 90:6B 91:CE 92:80 93:7E 94:73 95:00 96:85 97:50
98:90 99:69 9A:1D 9B:3E 9C:D0 9D:00 9E:7B 9F:6D
```

It's like the request to read starting from 0x88 is ignored and the subsequent read will return data from a random memory location.

lurch, 2021-11-17:
That indeed looks very odd. By looking at the values in your print-out, bytes 88-97 in your second "read" appear to match up with bytes EF-FE in your first read, and bytes 98-9F in your second read appear to match up with bytes 82-89 :confused: 
Is it possible that there's a loose wire or something, which is causing your sensor to randomly reset? (if you run that same program multiple times, do you always get identical output?)

I don't have a BMP280 myself, so I can't help diagnose this any further.

Glitchbone, 2021-11-17:
Yes if I run it multiple time, I always get the same output apart from the bytes related to the temperature / pressure registers.
So far I tried :
- Different BMP280 modules (confirmed working with an arduino UNO)
- Different wires / breadboard 
- Different i2c bus / pins
- Even a different Pico board :)
But with no luck...

And yes you're right, it seems that requesting bytes from 0x88 gives back bytes starting from 0xEF. That's so weird

Glitchbone, 2021-11-18:
**UPDATE**

The same setup works fine with CircuitPython on the same Pico board with the same wiring and BMP280 sensor.
I really don't understand...

lurch, 2021-11-18:
Have you tried running the i2c bus at a slower baudrate?

Glitchbone, 2021-11-18:
Nevermind I think I found the culprit...
Actually the example code is wrong.
The registers definitions should be declared as const uint8_t and not "defined". The current code is not passing the right value as the register address everytime it calls i2c_write_blocking(). I'm really no C expert but as I understand, (uint8_t*)0x88 (after the preprocessor replaced the value) passed as an argument to i2c_write_blocking() will result in the value at memory address 0x88 (of the Pico) after being dereferenced inside the function and not the actual literal value of 0x88. Am I right?

lurch, 2021-11-18:
ping @JamesH65 - IIRC this was one of the examples written by the interns?

JamesH65, 2021-11-19:
Oh dear, that code is properly borked. I vaguely remember having to make changes in there to make things all compile, but clearly didn't read the code well enough to do it properly, and I didn't have anything to test it on, so was just a compile test. I think the suggestion to make things const ints of the correct type would do the job, but might be quicker to just copy the defines to a local buffer and use that. I'll do a PR.

Glitchbone, 2021-11-19:
:)
The example is also present in the C SDK PDF doc that will need to be updated to reflect the changes (if it's not automated)

aallan, 2021-11-19:
> The example is also present in the C SDK PDF doc that will need to be updated to reflect the changes (if it's not automated)

It's automated.

JamesH65, 2021-11-19:
Typical - 2 minute change, 40 minutes of arsing around with git to get my trees up to date. Easier to re-clone the damn things than try and update.

JamesH65, 2021-11-19:
Please try this fix

 https://github.com/raspberrypi/pico-examples/pull/new/bme280_code_fix

If you can confirm if it works OK, I'll convert to a PR and ask for it to be merged.

Glitchbone, 2021-11-19:
I confirm that it works perfectly well now @JamesH65

JamesH65, 2021-11-19:
Excellent, will PR it.

JamesH65, 2021-11-19:
Fixed by https://github.com/raspberrypi/pico-examples/pull/187

closed 2021-11-19 by JamesH65

itopaloglu83, 2021-11-21:
Okay. It seems like https://github.com/raspberrypi/pico-examples/pull/187 introduced another error because `i2c_write_blocking` tries to unreferenced the given value and end up reading all 0x00 from the I2C bus. I'm creating another issue about this.

itopaloglu83, 2021-11-21:
I spent more time than I would like to admit diagnosing this issue on my own.

Thank you for the great fix!