raspberrypi/pico-examples #193: PWM, same slice, read and write, possible?

Day0Dreamer, 2022-01-09:
Hello there!

I'm tasked with putting 8 4-pin coolers on one Pico.
According to the Datasheet, there are 8 slices, where each slice has A and B channels.

Is it possible to utilize the same slice to generate and read in PWM signals?

From what I've gathered, it's not feasible because generation utilizes A and B channels both, whereas for reading you are required to use B channel.

Is this correct, or may there still be room to fit PWM gen and PWM read on the same slice, allowing to fit 8 fans?

Thank you.

lurch, 2022-01-12:
By "read in PWM signals" I assume you mean frequency-counting? I can't personally offer any more advice about PWM than what already appears in section "4.5. PWM" of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf but if you're using all the PWM slices for outputting PWM signals, you _might_ be able to measure the input frequency from your fans using PIO? See e.g. https://github.com/raspberrypi/pico-examples/tree/master/pio/logic_analyser and https://github.com/raspberrypi/pico-examples/tree/master/pio/quadrature_encoder  Or I guess you could run a simple GPIO-polling-loop on the second CPU core? Even 6000 RPM (i.e. 100Hz) is a fairly "slow" signal when the RP2040 is running at 125MHz :wink:

Day0Dreamer, 2022-01-16:
That is quite an idea. Not sure when I get around to it, but I have not considered actually using the 2nd core