raspberrypi/pico-examples #197: Pulse Position Modulation (PPM)

adaptivercsailing, 2022-02-08:
An invitation to use RPi Pico to develop RC (remote radio frequency control) pulse position modulation (PPM) waveforms. The RPi Zero and RPi Pico appear to be ideal candidates to create a new approach for providing alternate solutions to the **Adaptive RC Sailer Project** https://github.com/adaptivercsailing/adaptivercsailing. Request help duplicating the effects of the **Arduino Pulse Width Modulation Utility** https://github.com/ElectricRCAircraftGuy/eRCaGuy_PPM_Writer for use with RPi Zero and Pico.

elmot, 2023-08-20:
Hello @adaptivercsailing, would you like test my demo?
I have done that but I have no real hardware to test.
https://github.com/elmot/pico-examples/tree/master/pio/ppm

adaptivercsailing, 2023-10-08:
elmot,
I just saw your reply/offer from August 2023.
I'll start testing your code on Monday 10-09-2023.
Please correspond via dkender@wright.edu
Thank you - I am looking forward to working with you.
Dave K