raspberrypi/pico-examples #199: USB remote wakeup does not work  - dev_hid_composite

ThomasLeister, 2022-02-13:
Hi!,

I'm trying to use a Raspberry Pi Pico to wake up a PC (Fedora Linux) via the USB port. To check whether the remote wakeup feature is supported, I started the "dev_hid_composite" example on my Raspberry Pi Pico and attached it to my PC. To make sure the device remains powered on suspend, the pico is also powered via an external power supply (not only USB bus power). 

The dev_hid_composite example is working when the PC is in running state: It is detected as a HID device and I see key presses ("A") and mouse movements when I push the Pico's button. 

When the PC changes to "suspend" mode, the Pico continues flashing, but the frequency is lower (as expected), which means that USB suspend was detected correctly. 

Now, when I push the button, again, the Pico should wake up the host, according to https://github.com/raspberrypi/pico-examples/blob/master/usb/device/dev_hid_composite/main.c#L215

... but nothing happend. I've checked the settings on my PC: 

```
# grep . /sys/bus/usb/devices/**/power/wakeup
/sys/bus/usb/devices/1-1/power/wakeup:enabled
/sys/bus/usb/devices/1-3/power/wakeup:enabled
/sys/bus/usb/devices/usb1/power/wakeup:enabled
/sys/bus/usb/devices/usb2/power/wakeup:enabled
```

_TinyUSB device is `1-1`._

```
# lsusb -vv
[...]

...Bus 001 Device 027: ID cafe:4004 TinyUSB TinyUSB Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0xcafe 
  idProduct          0x4004 
  bcdDevice            1.00
  iManufacturer           1 TinyUSB
  iProduct                2 TinyUSB Device
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     239
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               5
...
```

The output shows that USB remote wakeup is available and enabled. Still, my computer is not reacting to any button presses and remains in suspend mode.  I've tried different ports and a modified USB cable, which does not transport power, just USB data + GND). Does anybody have a working USB remote wakeup setup or could verify that the feature works?

I'm a bit lost here ... but would be very happy to get this working :)

almastante, 2022-08-29:
Wakeup is working fine for me. I'm on Win 10. Loaded the example on Pico without any changes, put the PC to sleep, the led blinked slower. Pressing the button on Pico woke up the PC.

lurch, 2022-08-31:
ping @hathach

Slion, 2024-06-12:
Works well enough on Windows 11 too. Though if you just send HID reports is does not wake the machine. You need to send an explicit `tud_remote_wakeup();` apparently which is what the example does when suspended and pushing the board button.

nickovs, 2025-04-15:
@ThomasLeister Can you wake using a regular USB keyboard? If not it might be a BIOS setting issue.