raspberrypi/pico-examples #202: oled_i2c example works incorrectly on higher resolution displays

Addy771, 2022-03-06:
The oled_i2c example is written for a 128x32 pixel SSD1306 display. I tested the example with two SSD1306 displays with a resolution of 128x64. Both displays showed the same output when tested with the oled_i2c example. I was able to correct the display output with some changes.

Without modifying the example, the raspberries are drawn on the display on only half the lines:
[image]

I thought updating the OLED_HEIGHT #define to 64 to match my display's resolution would make an improvement, but it only introduced some randomly lit pixels:
[image]

I looked at the ssd1306 datasheet and experimented with different values for some of the parameters that are set in oled_init(). Changing the value sent for the OLED_SET_COM_PIN_CFG option from 0x02 to 0x12 (Alternative COM pin configuration) made a difference. The raspberries were now drawn as expected, but there were still randomly lit pixels below:

[image]

To fix this I changed the line creating a render_area struct to be based off the image height rather than the display height:
`    struct render_area area = {start_col: 0, end_col : IMG_WIDTH - 1, start_page : 0, end_page : (IMG_HEIGHT / OLED_PAGE_HEIGHT) - 1};`

Now the display looks correct:

[image]

I'm not sure whether the common ssd1306 OLED displays use sequential or alternative COM pin configuration, another #define could be added to allow switching between either mode.

lurch, 2022-03-06:
If you want to submit a PR that is fully backwards-compatible with the existing code then I guess we could think about including that. But we currently don't have any spare engineering resources to actively look at enhancing the example-code provided in this repo.

jderoo, 2022-04-24:
I ran into the same issues as @Addy771, although with a different 128x64 display:

[image]

And I came up with the same solutions. I did also have to change the OLED_ADDR from 0x3C to 0x3D. If I had looked here first I probably would have saved an hour or two, but I might have missed out on the joy of reading the data sheet!  So if anyone else is running into this, try Addy771's process above to modify your example code.

The changes to the #defines will need to be up to the user; in my code I flagged them.  I also wrapped the OLED_SET_COM_PIN_CFG in a conditional based on the value of the OLED_HEIGHT.  And the other change Addy771 made to the creation of a render_area struct fixed a latent bug in the original code that just happened to work with a 32 row display.  But Addy771's fix will work for either 32 or 64 rows, so it does not need to be conditionalized.  I can submit a PR, but I don't have the original display the example was written for to test it, so I can't really claim it would be "a PR that is fully backwards-compatible with the existing code".  But if someone wants to give it a try, let me know.

jderoo, 2022-04-25:
> I did also have to change the OLED_ADDR from 0x3C to 0x3D.

This was not completely correct.  I was messing around with another example/library and somehow got what I had to do for that one confused with this example.  This example masks the LSbit with a bitwise & 0xFE to zero the LSbit when doing a write.  To do a read it does the & 0xFF, which would only work if the 0x3D address were used.  However, there are no reads from the driver chip in this example, so that issue never shows up.