raspberrypi/pico-examples #204: Integer division is used instead of float.

rewolff, 2022-03-12:
In the WS2812 example, an integer division is used where a float was meant. 

```
diff --git a/pio/ws2812/ws2812.pio b/pio/ws2812/ws2812.pio
index 3c31fd6..75fdd48 100644
--- a/pio/ws2812/ws2812.pio
+++ b/pio/ws2812/ws2812.pio
@@ -39,7 +39,7 @@ static inline void ws2812_program_init(PIO pio, uint sm, uint offset, uint pin,
     sm_config_set_fifo_join(&c, PIO_FIFO_JOIN_TX);
 
     int cycles_per_bit = ws2812_T1 + ws2812_T2 + ws2812_T3;
-    float div = clock_get_hz(clk_sys) / (freq * cycles_per_bit);
+    float div = (float) clock_get_hz(clk_sys) / (freq * cycles_per_bit);
     sm_config_set_clkdiv(&c, div);
 
     pio_sm_init(pio, sm, offset, &c);

```

lurch, 2022-03-13:
Do you want to create a PR against the `develop` branch?

rewolff, 2022-03-14:
I understand the "ease" that this would create: Single click to merge. But this kind of stuff pollutes my own collection of repositories that I do actively work on. I'd prefer someone in charge just pastes the "(float)" into their sources and pushes commit. 

Where do I submit bugs (typos) in the RP2040 datasheet?

rewolff, 2022-03-14:
OK. Did it anyway.

aallan, 2022-03-14:
> Where do I submit bugs (typos) in the RP2040 datasheet?

That'd be the [pico-feedback](https://github.com/raspberrypi/pico-feedback/issues) repo.

kilograham, 2022-05-13:
um. that is actually a floating point division (as `freq` is a `float`). Can you check you are doing what you think you're doing. failing that i guess you could be using an ancient compiler?

rewolff, 2022-05-13:
Well... I actually did see the 156.000 when I expected 156.250. I just checked with the native compiler and it does the float division as you say it should. 
My arm-none-eabi-gcc says it is 7.3.1.

kilograham, 2022-05-16:
I tried it with 7.3.1 and it seems to work correctly. Can you post some sample code that shows the problem.

rewolff, 2022-05-17:
I think I was trying to do my own implementation of WS2812. That needed to run at 2.4MHz (800 kHz * 3) . I was not getting the results I wanted because I had forgotten the *3. But I didn't know that then. So I added a printf ("%f\n", div) and saw a divisor that was .000 instead of the .250 that I expected. 

The function is tagged inline, right?  This enables for example constant-propagation into the code that is being compiled. Could it be that gcc under specific circumstances forgets the "cast" of 800000 to float by the argument list?

I can't reproduce it now. Don't know what triggered it. I am actually convinced that when I reported this, I though the compiler was absolutely right in doing what it did. And I knew there was a float in the RHS, and the promotion to float calculations was happening AFTER the integer division. 

Sorry. lets write this down to user error and close it.

closed 2022-05-17 by rewolff