raspberrypi/pico-examples #206: Minor typo in IR NEC PIO setup

hamuelbones, 2022-04-12:
While debugging IR communication on a custom board, I came across a discrepancy between the comment and actual clock divider setup in `nec_receive.pio`:

```
    // Set the clock divider to 10 ticks per 562.5us burst period
    //
    float div = clock_get_hz (clk_sys) / (10.0 / 526.6e-6);
    sm_config_set_clkdiv (&c, div);
```

Note the comment says it wants to set the clock divider so that it clocks 10 times in 5**62.5**us, but the comment is actually setting it so it is 10 ticks in 5**26.6**us. My understanding is that the comment is right in describing what should happen and that the code is setting the clock divider to a value that is slightly too high. The transmit code uses 562.5us in comments and clock divider settings properly already.

In my testing both values work, and from reading the code it looks like there would be some tolerance to the clock frequency being a little bit fast. But it is probably better to make the clock divider be what it was intended, unless I'm misunderstanding something (I'm by no means an expert at using PIO... yet :))

Will raise a pull request shortly.

lurch, 2022-04-12:
ping @mjcross (the original author of this code).

mjcross, 2022-04-12:
Well spotted - yes that's a mistake on my part

Mostafa-wael, 2022-05-03:
Can I work on this issue ?

lurch, 2022-05-03:
> Can I work on this issue ?

Huh? I think it has already been fixed by #207 ?

Mostafa-wael, 2022-05-03:
But, the PR wasn't merged and the issue is still open. 
What do you think ?

Mostafa-wael, 2022-05-03:
Okay, thanks for your response

closed 2022-05-16 by hamuelbones