raspberrypi/pico-examples #208: Example request: show how to use pio_sm_set_clkdiv()

beroset, 2022-05-04:
I am trying to use the `pio_sm_set_clkdiv()` call in my code and had some questions about it.  I had hoped to find an example of its use in the examples, but currently there are none.  It might be useful to add it, perhaps as an option to the squarewave example, just to show how it works and how it is intended to be used.

lurch, 2022-05-04:
There _are_ [examples](https://github.com/raspberrypi/pico-examples/search?q=sm_config_set_clkdiv) of `sm_config_set_clkdiv`, and AIUI the difference is largely that you'd use `sm_config_set_clkdiv` to configure the StateMachine before starting it, whereas `pio_sm_set_clkdiv` would be used to modify the clock divider (see section 3.5.5 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf ) of a StateMachine that's already running?

lurch, 2022-05-04:
Sorry for the inaccuracy in my earlier statement; `pio_sm_set_clkdiv` isn't _only_ useful after the state-machine is running, it's just that the common usage-model in `pico-sdk` is to use the `sm_config` methods to set up the state-machines _before_ you start them running (which IIRC is more efficient than using the `pio_sm` methods).

And that's just about the limit of my knowledge on this subject, so any further clarifications will have to come from @kilograham