raspberrypi/pico-examples #223: lcd_1602_i2c not building on PI 4

jksevend, 2022-06-23:
Hello guys :) 

I have been using a Raspberry Pi to manage my pico projects and recently wanted to try out the lcd example.
I used the pico setup shell script from the official c++ sdk documentation pdf and the blink example worked.
However this particular example does not. 
The error I get during building is: 
```
fatal error: hardware/i2c: No such file or directory
 #include "hardware/i2c"
          ^~~~~~~~~~~~~~
compilation terminated.
```

My ``CMakeLists.txt``:
```
cmake_minimum_required(VERSION 3.12)

set(PICO_SDK_PATH $ENV{PICO_SDK_PATH})
# Pull in SDK (must be before project)
include(${PICO_SDK_PATH}\\pico_sdk_init.cmake)

project(pico-lcd-temperature C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

add_executable(${PROJECT_NAME} main.cpp)

# pull in common dependencies
target_link_libraries(${PROJECT_NAME} pico_stdlib hardware_i2c)

# create map/bin/hex file etc.
pico_add_extra_outputs(${PROJECT_NAME})
```

And there is basically only one cpp file - the content of it was pasted from your [example repo](https://github.com/raspberrypi/pico-examples/tree/master/i2c/lcd_1602_i2c)

Been scratching my head why it will not build. Anyone have an idea how to fix this?

lurch, 2022-06-23:
Your error says:
```
#include "hardware/i2c"
```
but [lcd_1602_i2c.c](https://github.com/raspberrypi/pico-examples/blob/master/i2c/lcd_1602_i2c/lcd_1602_i2c.c#L10) says:
```
#include "hardware/i2c.h"
```
?

jksevend, 2022-06-24:
Yes exactly. I thought that maybe im missing the built hardware library but i guess the pico setup script installs every needed library.

Also i found out that i can build the cloned lcd example. Just not in my own project

lurch, 2022-06-24:
> Also i found out that i can build the cloned lcd example. Just not in my own project

Yes, because it looks like you've missed the `.h` off the end of the `#include` line, as I pointed out in my previous reply?

jksevend, 2022-06-24:
Oh my... this is embarassing. It works now. I am sorry.

closed 2022-06-24 by jksevend